riscv-software-src/opensbi #67: Support M-mode to M-mode traps (avoid stack overwrite)

xerpi, 2019-02-18:
At init both the `sp` and `CSR_MSCRATCH` registers are set to the same value (some offset from `tp`).

Then, when entering `_trap_handler` from M-mode (instead of S-mode), the `sp` will be set to `CSR_MSCRATCH` which is the base of the M-mode stack, and therefore the M-mode context stack before the trap will be overwritten.

I guess the solution would be to use a totally distinct stack spaces for both `sp` and `CSR_MSCRATCH`.

avpatel, 2019-02-19:
Another way of doing this is to not update SP in _trap_handler when previous mode was M-mode.

xerpi, 2019-02-19:
I'm trying to implement that but it turns out to be more difficult than expected since OpenSBI relies on having a valid `mscratch` (used for example by `sbi_scratch_thishart_ptr()`, unlike Linux, which doesn't use that register in the kernel and just sets it to `0`  when entering an exception (https://github.com/torvalds/linux/blob/master/arch/riscv/kernel/entry.S#L158).

avpatel, 2019-02-19:
Here you go https://github.com/riscv/opensbi/pull/68

xerpi, 2019-02-19:
Nice! Many thanks for the fast responses and the PR :)

avpatel, 2019-02-20:
The PR is merged now hence closing this issue.

closed 2019-02-20 by avpatel