riscv-software-src/opensbi #81: Fails to compile with gcc 9.0

kraj, 2019-03-06:
see https://errors.yoctoproject.org/Errors/Details/231564/

alistair23, 2019-03-06:
Interesting, GCC9 is pointing out something which is a bad idea to have in code. For RISC-V it shouldn't cause catastrophic problems, but will slow things down. Even then our struct should always be aligned, so this really won't cause us any issues for us.

For more details you can read the GCC bug report that started this: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=51628

I'll work on a patch, I think an intermediate variable will fix it, as that way we never take an address of a packed struct. A little bit of a pain, but I don't see a much better solution.

alistair23, 2019-03-07:
@kraj Do you have a branch with GCC9 that I can use?

kraj, 2019-03-07:
Use Yoe Distro ( shameless plug ) which I co-maintain and use yoe/mut branch
https://github.com/YoeDistro/yoe-distro

or you can use yoe/mut branch from my contrib repo
https://github.com/YoeDistro/openembedded-core/tree/yoe/mut

and in `local.conf` set

```
GCCVERSION = "9.%"
```

kraj, 2019-03-07:
> Interesting, GCC9 is pointing out something which is a bad idea to have in code. For RISC-V it shouldn't cause catastrophic problems, but will slow things down. Even then our struct should always be aligned, so this really won't cause us any issues for us.
> 

I think we should avoid typecasting packed structs to pointers, or not pack them

> For more details you can read the GCC bug report that started this: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=51628
> 

Yes I am aware of that

> I'll work on a patch, I think an intermediate variable will fix it, as that way we never take an address of a packed struct. A little bit of a pain, but I don't see a much better solution.

that will be awesome

closed 2019-03-10 by avpatel