riscv-software-src/opensbi #85: Code logic problem (coldboot/warmboot)

wxjstz, 2019-03-07:
```c
static atomic_t coldboot_lottery = ATOMIC_INITIALIZER(0);

void __noreturn sbi_init(struct sbi_scratch *scratch)
{
	bool coldboot = FALSE;
	u32 hartid = sbi_current_hartid();
	const struct sbi_platform *plat = sbi_platform_ptr(scratch);

	if (sbi_platform_hart_disabled(plat, hartid))
		sbi_hart_hang();
		
	if (atomic_add_return(&coldboot_lottery, 1) == 1)
		coldboot = TRUE;

	if (coldboot)
		init_coldboot(scratch, hartid);
	else
		init_warmboot(scratch, hartid);
}
```

These codes cannot determine **coldboot** or **warmboot**.

These codes can only be used to select one processor to perform common tasks.

What is the true meaning of these codes?

If it is **coldboot/warmboot**, `coldboot_lottery` should not be a static variable, it should be a reserved memory, because the program will be reloaded under normal circumstances, the value will be reset when warmboot.

If it is select a processor to perform common tasks. The naming of these codes is strange.

avpatel, 2019-03-07:
To be more semantically correct, there are two types of init:
1. full init (i.e. initialize everything from scratch including some platform specific peripherals)
2. lite init (i.e. initialize only current HART and related platform specific peripherals)

The "full init" is supposed to be done only once whereas "lite init" is done every time a HART is powered-up. In future, when we will have HART hotplug the "lite init" will be called every time a HART is powered-up at runtime.

Now, we have been using "coldboot" term to represent "full init" and "warmboot" term to represent "lite init".

I think "coldboot" and "warmboot" can little confusing if we start thinking from HW power-management sequences.

Let's use more SW terms for "full init" and "lite init". Do you have better suggestions?

wxjstz, 2019-03-07:
opensbi will be reloaded each time the motherboard is reset. For example, on HiFive Unleashed, the program is copied from the SD card to the memory. So the `coldboot_lottery` variable is always 0 at reset. So, whether it is a warm boot or a cold boot will execute coldboot.

coldboot_lottery should be placed outside of the executable image, and atomic swap instructions should be executed here instead of atomic additions(In extreme cases, the addition will overflow).

```c
/* coldboot memory is a random value, it is impossible to 0x012345678 */
If (__sync_lock_test_and_set(&coldboot_lottery,0x012345678) != 0x012345678)
	coldboot()
else
	warmboot()
```

avpatel, 2019-03-08:
You suggestion looks good to me.

May be use "#define" for 0x12345678 (32bit magic const). Please send a PR. If possible also test on Unleashed board few times.

wxjstz, 2019-03-08:
Thank you very much!!!

I am still confused.
1. Is it necessary to perform a `coldboot` after a soft reset? 
2. Where is the entry of soft reset?

avpatel, 2019-03-08:
I think let's rename "coldboot" to "full_init" and "warmboot" to "lite_init". This will make it more explicit about SW code path. Some of the platform callbacks have "cold_boot" parameter which should be renamed too.

The booting stages prior to OpenSBI runtime firmware will take-care of differing HW sequence for "coldboot" and "warmboot".

The entry for "lite init" is specified in "struct sbi_scratch" as "warmboot_addr" member. For OpenSBI firmware, the "lite init" entry point is _start_warm() and "full init" entry point is _start().

If you have better names then please suggest.

avpatel, 2019-03-08:
Also, generally runtime firmware don't need to distinguish between HW "coldboot" and "warmboot". The power-managment controller or on-chip ROM firmware will differentiate between HW "coldboot" and "warmboot".

A runtime firmware certainly needs to have two types of init such as "full init" and "lite init". It also needs separate low-level entry point "lite init" to support CPU/HART hotplug.

Due above I suggest renaming "coldboot" and "warmboot".

wxjstz, 2019-03-08:
Thank you very much! 

I think the naming is OK, just add some comments.

closed 2019-03-08 by wxjstz