riscv-software-src/opensbi #97: plic: invalid register write: 000001fc

davidlt, 2019-03-19:
I am running QEMU emulator version 3.1.50 (qemu-3.1.50.201903180749.1d023a5-1.fc29), QEMU commit 1d023a5.

I am using master of OpenSBI, I get the following:
```
qemu-system-riscv64: plic: invalid register write: 000001fc

OpenSBI v0.3 (Mar 19 2019 08:32:46)
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name          : QEMU Virt Machine
Platform HART Features : RV64ACDFIMSU
Platform Max HARTs     : 8
Current Hart           : 1
Firmware Base          : 0x80000000
Firmware Size          : 100 KB
Runtime SBI Version    : 0.1
```

It boots and it works just fine.

alistair23, 2019-03-19:
I'm not seeing this on the latest QEMU, can you try with today's master and let me know if you still see it?

davidlt, 2019-03-20:
I am on QEMU master now (which is also rc0):
```
QEMU emulator version 3.1.90 (qemu-3.1.90.201903200002.62a172e-1.fc29)
```

I still get `qemu-system-riscv64: plic: invalid register write: 000001fc`

I am using 5.1 kernel, GCC 9, binutils 2.32 and OpenSBI was built native on riscv64.

My kernel is ELF (switching to Image.gz now) and thus I did what BBL does:
```
objcopy -O binary --set-section-flags .bss=alloc,load,contents
```

alistair23, 2019-03-20:
Yep, I can reproduce this. I have a fix for QEMU, I'll send the series out today after my tests finish.

I also have a change to OpenSBI which updates the code to match the documentation, it is here: https://github.com/riscv/opensbi/pull/98

alistair23, 2019-03-21:
You can see the branch here: https://github.com/alistair23/qemu/tree/mainline/alistair/plic.next

The patches are on the list

avpatel, 2019-04-10:
The issue is now fixed in latest QEMU hence closing this issue.

closed 2019-04-10 by avpatel

wzis, 2019-12-09:
I'm running qemu 4.2, and have a fedora running on qemu-system-riscv64, using:
$ qemu-system-riscv64 \
  -nographic \
  -machine virt \
  -smp 8 \
  -m 2G \
  -kernel Fedora-Developer-Rawhide-*-fw_payload-uboot-qemu-virt-smode.elf \
  -object rng-random,filename=/dev/urandom,id=rng0 \
  -device virtio-rng-device,rng=rng0 \
  -device virtio-blk-device,drive=hd0 \
  -drive file=Fedora-Developer-Rawhide-*.raw,format=raw,id=hd0 \
  -device virtio-net-device,netdev=usernet \
  -netdev user,id=usernet,hostfwd=tcp::10000-:22

However, after the machine is up, I can't use the port 10000 on host to ssh to it：
# ssh -p 10000 root@localhost

I have put the public key generated on the host to the VM's ~root/.ssh/authorized_keys. But found the openssh server is not working as expected: even from the VM, when I try to run “ssh riscv@localhost", it will not work.