riscv-software-src/opensbi #110: Publish RISC-C OpenSBI spec somewhere in OpenSBI repo

changab, 2019-04-03:
In order to standardize the incorporation of OpenSBI on RISC-V UEFI port in UEFI Platform Initialization spec, we need to public RISC-C OpenSBI spec somewhere in OpenSBI repo.  So we can put the external references to RISC-V OpenSBI repo as the reference of sbi_scratch and sbi_platform structure in PI spec.

alistair23, 2019-05-28:
What do you mean by the OpenSBI spec?

Shouldn't the only interface with OpenSBI be using SBI calls?

changab, 2019-05-31:
Yes. UEFI edk2 open source uses SBI calls. I mean we will mention OpenSBI in UEFI spec and put the external references in UEFI spec which point to OpenSBI spec, sbi_scratch, platform implementation guidance of sbi_platform and etc. The OpenSBI spec refers to "RISC-V OpenSBI v0.3.pdf". I forget how and where I got it though.

atishp04, 2019-05-31:
@changab It is available at https://github.com/riscv/opensbi/releases 
Every release tar ball will contain the entire binary & RISC-V OpenSBI vX.Y.pdf

changab, 2019-05-31:
Thanks, Atish!

atishp04, 2019-06-08:
closing this issue.

closed 2019-06-08 by atishp04