riscv-software-src/opensbi #118: it cannot load the kernel

figozhang, 2019-04-24:
i am using linux 5.0 kernel, but it canot load the kernel on qemu, what is the problem?

figo@figo-OptiPlex-9020: linux 5.0$ qemu-system-riscv64 -machine virt -kernel opensbi/build/platform/qemu/virt/firmware/fw_jump.elf -device loader,file=arch/riscv/boot/Image,addr=0x80200000 -nographic --append "rdinit=/linuxrc console=ttyAMA0"
qemu-system-riscv64: plic: invalid register write: 000001fc

OpenSBI v0.3 (Apr 25 2019 05:45:54)
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name          : QEMU Virt Machine
Platform HART Features : RV64ACDFIMSU
Platform Max HARTs     : 8
Current Hart           : 0
Firmware Base          : 0x80000000
Firmware Size          : 100 KB
Runtime SBI Version    : 0.1

PMP0: 0x0000000080000000-0x000000008001ffff (A)
PMP1: 0x0000000000000000-0xffffffffffffffff (A,R,W,X)

atishp04, 2019-04-24:
I just verified 5.0 kernel. It works. I have used the exact command line as per docs (https://github.com/riscv/opensbi/blob/master/docs/platform/qemu_virt.md)

I think issue is console=ttyAMA0. It should be console=ttyS0

atishp04, 2019-04-29:
@figozhang : Can we close this issue if it is resolved for you ?

closed 2019-05-01 by avpatel