riscv-software-src/opensbi #122: cannot reboot the system

figozhang, 2019-04-29:
i am using Linux 5.0 kernel and opensbi, i found it cannot reboot:

root@benshushu:~#
root@benshushu:~#
root@benshushu:~# reboot
[  OK  ] Closed LVM2 poll daemon socket.
[  OK  ] Stopped target Timers.
[  OK  ] Stopped Daily apt upgrade and clean activities.
[  OK  ] Stopped target Graphical Interface.
[  OK  ] Stopped Daily rotation of log files.
[  OK  ] Stopped target Multi-User System.
         Stopping Regular background program processing daemon...
[  OK  ] Stopped Periodic ext4 Onli…ata Check for All Filesystems.
[  OK  ] Stopped Daily apt download activities.
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped target Login Prompts.
         Stopping Getty on tty1...
[  OK  ] Stopped getty on tty2-tty6… and logind are not available.
         Stopping Getty on tty5...
         Stopping Serial Getty on hvc0...
         Stopping Getty on tty3...
         Stopping Getty on tty4...
         Stopping Getty on tty6...
         Stopping Getty on tty2...
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
         Stopping Serial Getty on ttyS0...
         Stopping Availability of block devices...
[  OK  ] Stopped Regular background program processing daemon.
[  OK  ] Stopped Getty on tty2.
[  OK  ] Stopped Serial Getty on hvc0.
[  OK  ] Stopped Getty on tty1.
[  OK  ] Stopped Serial Getty on ttyS0.
[  OK  ] Stopped Getty on tty3.
[  OK  ] Stopped Getty on tty4.
[  OK  ] Stopped Getty on tty5.
[  OK  ] Stopped Getty on tty6.
[  OK  ] Removed slice system-serial\x2dgetty.slice.
[  OK  ] Removed slice system-getty.slice.
         Stopping Permit User Sessions...
[  OK  ] Stopped Availability of block devices.
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped target Network.
         Stopping ifup for eth0...
         Stopping Raise network interfaces...
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Slices.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target Sockets.
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target System Initialization.
         Stopping Load/Save Random Seed...
[  OK  ] Stopped target Swap.
[  OK  ] Stopped target Local Encrypted Volumes.
[  OK  ] Stopped Forward Password R…uests to Wall Directory Watch.
[  OK  ] Stopped Dispatch Password …ts to Console Directory Watch.
         Stopping Update UTMP about System Boot/Shutdown...
         Stopping Network Time Synchronization...
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Raise network interfaces.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped ifup for eth0.
[  OK  ] Stopped Apply Kernel Variables.
[  OK  ] Stopped Load Kernel Modules.
[  OK  ] Stopped target Local File Systems.
         Unmounting /mnt...
[  OK  ] Unmounted /mnt.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Create System Users.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[  OK  ] Reached target Final Step.
[  OK  ] Started Reboot.
[  OK  ] Reached target Reboot.
         Stopping Monitoring of LVM…meventd or progress polling...
[   64.117212] printk: systemd-shutdow: 28 output lines suppressed due to ratelimiting
[   64.997467] systemd-shutdown[1]: Syncing filesystems and block devices.
[   65.155286] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   65.214518] systemd-journald[86]: Received SIGTERM from PID 1 (systemd-shutdow).
[   65.239998] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   65.296737] systemd-shutdown[1]: Unmounting file systems.
[   65.314410] [265]: Remounting '/' read-only in with options '(null)'.
[   65.438658] EXT4-fs (vda): re-mounted. Opts: (null)
[   65.460945] systemd-shutdown[1]: All filesystems unmounted.
[   65.461367] systemd-shutdown[1]: Deactivating swaps.
[   65.462366] systemd-shutdown[1]: All swaps deactivated.
[   65.462689] systemd-shutdown[1]: Detaching loop devices.
[   65.498962] systemd-shutdown[1]: All loop devices detached.
[   65.499464] systemd-shutdown[1]: Detaching DM devices.
[   65.536748] reboot: Restarting system
[   86.532464] rcu: INFO: rcu_sched self-detected stall on CPU
[   86.533065] rcu:     0-....: (5249 ticks this GP) idle=702/1/0x4000000000000002 softirq=16275/16275 fqs=2614
[   86.533453] rcu:      (t=5250 jiffies g=9509 q=18)
[   86.533891] Task dump for CPU 0:
[   86.534165] systemd-shutdow R  running task        0     1      0 0x00000008
[   86.534537] Call Trace:
[   86.534754] [<ffffffe00005345e>] walk_stackframe+0x0/0x156
[   86.535037] [<ffffffe000053620>] show_stack+0x30/0x3a
[   86.535292] [<ffffffe0000bdf8a>] sched_show_task+0x408/0x41e
[   86.535564] [<ffffffe0000c120a>] dump_cpu_task+0x58/0x62
[   86.535824] [<ffffffe00013cb6e>] rcu_dump_cpu_stacks+0x9e/0x110
[   86.536182] [<ffffffe00013d9a2>] print_cpu_stall+0x1f0/0x438
[   86.536660] [<ffffffe00013e0e8>] check_cpu_stall+0x4fe/0x594
[   86.536930] [<ffffffe000144b2a>] rcu_pending+0x46/0x28c
[   86.537183] [<ffffffe0001432be>] rcu_check_callbacks+0x206/0x224
[   86.537464] [<ffffffe0001548ec>] update_process_times+0x30/0x110
[   86.537737] [<ffffffe00016d978>] tick_periodic+0x202/0x212
[   86.537993] [<ffffffe00016d9b0>] tick_handle_periodic+0x28/0x98
[   86.538264] [<ffffffe0008f824e>] riscv_timer_interrupt+0x50/0x5a
[   86.538582] [<ffffffe000d17b64>] do_IRQ+0xd4/0x1a2
[   86.538869] [<ffffffe000050900>] ret_from_exception+0x0/0xc
[   86.539143] [<ffffffe000051eae>] machine_restart+0x14/0x16

avpatel, 2019-05-06:
Currently, reboot/shutdown support is pending for SiFive Unleashed Platform in OpenSBI. Can you send patch?

closed 2023-01-12 by atishp04

BlackS52, 2025-07-30:
I have the same problem
Do we have any tool to reboot Lichee Pi4A board ?