riscv-software-src/opensbi #123: Building opensbi with Debian gcc-riscv64-linux-gnu fails to load

carlosedp, 2019-04-30:
OpenSBI fails to load Kernel (or loading fw_payload.elf) when built with gcc-riscv64-linux-gnu from https://packages.debian.org/buster/gcc-riscv64-linux-gnu installed with:

```bash
sudo dpkg --add-architecture riscv64
sudo apt-get install gcc-riscv64-linux-gnu g++-riscv64-linux-gnu
sudo sh -c "cat >/etc/ld.so.conf.d/riscv64-linux-gnu.conf <<EOF
/usr/local/lib/riscv64-linux-gnu
/lib/riscv64-linux-gnu
/usr/lib/riscv64-linux-gnu
/usr/riscv64-linux-gnu/lib/
EOF
"

riscv64-linux-gcc --version
riscv64-linux-gcc.br_real (Buildroot 2018.11-rc2-00003-ga0787e9) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

When building with Bootlin (https://toolchains.bootlin.com/downloads/releases/toolchains/riscv64/tarballs/riscv64--glibc--bleeding-edge-2018.11-1.tar.bz2) riscv64--glibc--bleeding-edge-2018.11-1.tar.bz2 it succeeds.

avpatel, 2019-04-30:
This is now fixed in latest OpenSBI.

The commit that fixes it is:, https://github.com/riscv/opensbi/commit/f9643f3472199f01e8eecdd6d50a379989646f03

carlosedp, 2019-04-30:
Just confirmed it works:

```
➜ qemu-system-riscv64 -M virt -m 256M -nographic -kernel fw_jump.elf

OpenSBI v0.3 (Apr 30 2019 11:38:10)
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name          : QEMU Virt Machine
Platform HART Features : RV64ACDFIMSU
Platform Max HARTs     : 8
Current Hart           : 0
Firmware Base          : 0x80000000
Firmware Size          : 100 KB
Runtime SBI Version    : 0.1

PMP0: 0x0000000080000000-0x000000008001ffff (A)
PMP1: 0x0000000000000000-0xffffffffffffffff (A,R,W,X)
QEMU: Terminated

➜ qemu-system-riscv64 -M virt -m 256M -nographic -kernel fw_payload.elf

OpenSBI v0.3 (Apr 30 2019 11:38:10)
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name          : QEMU Virt Machine
Platform HART Features : RV64ACDFIMSU
Platform Max HARTs     : 8
Current Hart           : 0
Firmware Base          : 0x80000000
Firmware Size          : 104 KB
Runtime SBI Version    : 0.1

PMP0: 0x0000000080000000-0x000000008001ffff (A)
PMP1: 0x0000000000000000-0xffffffffffffffff (A,R,W,X)

Test payload running
QEMU: Terminated
```

Thanks for the fast fix!

closed 2019-04-30 by carlosedp