riscv-software-src/opensbi #128: MSIP is supposed to be read-only in MIP?

taylor-bsg, 2019-05-26:
https://github.com/riscv/opensbi/blob/a6395acd6cb2c35871481d3e4f0beaf449f8c0fd/lib/sbi_hart.c#L351

riscv-privileged-20190405: The mip register is an MXLEN-bit read/write register containing information on pending interrupts, while mie is the corresponding MXLEN-bit read/write register containing interrupt enable bits. Only the bits corresponding to lower-privilege software interrupts (USIP, SSIP), timer interrupts (UTIP, STIP), and external interrupts (UEIP, SEIP) in mip are writable through this CSR address; the remaining bits are read-only. 

Is this targeting older, non-compliant HW, or am I misunderstanding something?

alistair23, 2019-05-28:
OpenSBI runs in M-mode, so it is able to access the M mode bits. See the end of that paragraph: 

>  A hart can write its own MSIP bit using the same memorymapped control register.

taylor-bsg, 2019-05-28:
Yes, that means that you need to write to the memory location, not use the
csr write instruction.

alistair23, 2019-05-28:
Ah yeah, it looks like you are correct. You should send a PR to the repo.

taylor-bsg, 2019-05-28:
I guess that means that this works on Spike, SiFive silicon, or Rocket?
What are you testing on? Maybe we should file an issue on those repos.

alistair23, 2019-05-28:
We test mostly on QEMU, HiFive UnLeashed and the Kendryte board.

taylor-bsg, 2019-05-28:
Thanks.   We will eventually submit a pull request, but others are free to
submit it before us.

avpatel, 2019-09-28:
We have taken care of this issue hence closing it.

closed 2019-09-28 by avpatel