riscv-software-src/opensbi #129: Use spaces replace Tab in source code?

changab, 2019-05-30:
Shall we use space in the source code instead of using Tab? The code look in the same indent in the different editors.

atishp04, 2019-05-30:
I think Linux coding guidelines suggests the indention should be set to 8 characters.
As long as the editor has 8 character tab settings, it shouldn't matter.

avpatel, 2019-05-31:
@changab We are following Linux coding style so that people familiar with Linux, U-Boot, and other projects using this style can easily contribute. I think it should easy for anyone to configure their editor for Linux coding style.

changab, 2019-06-01:
@avpatel I understand the consideration of Linux coding standard. However, OpenSBI is not just only for Linux programmer now, it will be also part of ekd2 and edk2 has it's own coding standard as well. Not insist it, I just think maybe a compromise could make different development worlds feel easy when coding. They don't have to change settings in the editor. Especially to work on two coding standards in one editor.
Thanks

marcopeereboom, 2019-06-03:
the 90s called and they want their editor wars back

closed 2019-06-29 by changab