riscv-software-src/opensbi #131: Use cpu_to_fdt32 instead of fdt32_to_cpu?

Georgeto, 2019-07-22:
https://github.com/riscv/opensbi/blob/3048f979cab3d0f42461bd4d96a4bdecf1791f62/lib/utils/irqchip/plic.c#L71

Shouldn't this be `cpu_to_fdt32(0xffffffff)`? (although it does not make a difference in this particular instance)

avpatel, 2019-07-24:
Yes, it should be cpu_to_fdt32() for being semantically correct.

It's a typo which slipped through cracks previously.

Please send patch to OpenSBI mailing list for correcting this.

Thanks,
Anup

avpatel, 2019-07-24:
I forgot to mention.

You can join OpenSBI mailing list using following link:
http://lists.infradead.org/mailman/listinfo/opensbi

Georgeto, 2019-07-24:
I posted a patch on the OpenSBI mailing list, and also subscribed to it.

avpatel, 2019-07-24:
Yes, got your patch on list.

I will merge it tomorrow.

Thanks,
Anup

avpatel, 2019-07-25:
Applied your patch.

Thanks,
Anup

closed 2019-07-25 by avpatel