riscv-software-src/opensbi #132: Is Floating Point Required?

Jesse-Millwood, 2019-07-24:
Is there a requirement for a floating point unit? I tried compiling with 

```
make CROSS_COMPILE=riscv64-unknown-elf- \
       PLATFORM_RISCV_XLEN=32 \
       PLATFORM_RISCV_ABI=ilp32 \
       PLATFORM_RISCV_ISA=rv32imac \
       PLATFORM=qemu/virt \
       V=1 \
       O=./build/
```
It seems I get the error in `riscv_fp.h` and the `#define` guards look like it is required.

avpatel, 2019-07-24:
We wanted to support misaligned load/store emulation for both general registers and float registers but it is quite possible to have a system without float support.

Most platforms we started with initially had floating point so riscv_fp.h has a compile time check.

I do agree that we should not be mandating float any more.

Let me explore this more.

dpetrisko, 2019-07-24:
We are also interested in compiling without FP (for now we just comment out irrelevant code, but it would be nice to have an officially supported configuration)

Jesse-Millwood, 2019-07-24:
Great, thanks for the reply. So I wasn't misconfiguring anything then. 

Looks like if that `#else` that triggers that `#error` in that file are removed and the `#endif` moved to the end, then the `sbi_misaligned_ldst.c` has an error because the `fp` variable must get set and make it okay to call those `{SET,GET}_F*_*` macros. 

I'm not entirely sure how the `mepc` is being used there. Could you explain that?

Jesse-Millwood, 2019-07-24:
> We are also interested in compiling without FP (for now we just comment out irrelevant code, but it would be nice to have an officially supported configuration)

Would you happen to have a patch of all that you needed to comment out to help drive what guards and configuration variables would be needed?

dpetrisko, 2019-07-24:
> Would you happen to have a patch of all that you needed to comment out to help drive what guards and configuration variables would be needed?

We haven't booted using OpenSBI yet, just gotten it to compile.  I'll make sure to followup if we get it working (or make a PR if it seems generally useful :))

atishp04, 2019-07-24:
@dpetrisko : Just curious which platform/RISC-V core are you trying to boot ?

Jesse-Millwood, 2019-07-25:
> I'm not entirely sure how the `mepc` is being used there. Could you explain that?

Actually I get it now, you're trying figuring out what kind of load instruction it is. For some reason I thought you were using it to figure out if floating point is supported or something but it makes sense now.

avpatel, 2019-07-25:
There is one more case where users can run OpenSBI firmware compiled without FP on HW with FP. In this case, OpenSBI firmware should blindly redirect FP related misaligned load/store traps to S-mode.

I have send a patch to OpenSBI mailing list which address FP compilation issue and above mentioned case. Please try this patch at your end.

You can join OpenSBI mailing list using following link:
http://lists.infradead.org/mailman/listinfo/opensbi

Regards,
Anup

avpatel, 2019-08-05:
I am closing this now. Please re-open it if not resolved.

closed 2019-08-05 by avpatel