riscv-software-src/opensbi #137: Is it possible to use vmlinux with opensbi?

joshuaahklein, 2019-09-26:
I've read that bbl is/is going to be deprecated, so I am trying to move to a more current bootloader.  I use riscv64 QEMU with the stage4 disk image and have been attempting to load a vmlinux kernel with opensbi with no success (note I am able to use the normal kernel image with opensbi, but I specifically need to use a statically-linked linux build).

avpatel, 2019-09-26:
We have different firmware types. One of these firmware FW_PAYLOAD is equivalent to what BBL does (i.e. takes Linux kernel as payload). The FW_PAYLOAD takes flat kernel image as payload (not vmlinux ELF). The BBL on other hand takes vmlinux ELF and converts it into flat kernel image before using as payload. In other words, OpenSBI FW_PAYLOAD and BBL deal with payload in same way.

We have other interesting firmware too such as FW_JUMP and FW_DYNAMIC. The FW_JUMP will be
default bios for QEMU RISC-V from QEMU 4.2 release onwards. The FW_DYNAMIC is going to be used with U-Boot and Coreboot bootloaders.

To start playing with OpenSBI on QEMU, I suggest you refer:
<opensbi_source>/docs/platform/qemu_virt.md

avpatel, 2019-09-27:
I will wait for 1 more day. If there are no more comments then I will close this issue.

joshuaahklein, 2019-09-27:
Thanks for the clarification.  I think I'm all set now.

closed 2019-09-27 by joshuaahklein