riscv-software-src/opensbi #139: The values of CAUSE_HYPERVISOR_ECALL and CAUSE_SUPERVISOR_ECALL macro

knknkn1162, 2019-10-21:
Hi.

I noticed that the values of CAUSE_HYPERVISOR_ECALL and CAUSE_SUPERVISOR_ECALL were actually reversed. Is it right?

https://github.com/riscv/opensbi/blob/be92da280d87c38a2e0adc5d3f43bab7b5468f09/include/sbi/riscv_encoding.h#L472-L473

Whereas, the [Privileged ISA Specification; 20190608version](https://riscv.org/specifications/privileged-isa/) says below:

[image]

avpatel, 2019-10-21:
Maybe the #define name are little miss-leading but when H-extension is available:
0x9 => Environment call from HS-mode
0xa => Environment call from VS-mode

The rationale is that when H-extension is available, the traditional S-mode is actually HS-mode
with hypervisor capabilities.

All ECALLs from VS-mode should be handled in HS-mode only hence we are delegating CAUSE_SUPERVISOR_ECALL to HS-mode when H-extension is available.
(Refer, delegate_traps() in opensbi/lib/sbi/sbi_hart.c)

I hope above info helps.

Regards,
Anup

knknkn1162, 2019-10-21:
I see.. Unfortunately, I'm currently not very familiar with H(Hypervisor)-mode, but I understood what you comment. Thank you!

avpatel, 2019-10-22:
Closing this issue now.

closed 2019-10-22 by avpatel