riscv-software-src/opensbi #140: fw_payload.o(.text+0x1961): 15 bytes required for alignment to 16-byte boundary, but only 14 present

davidlt, 2019-10-22:
I have updated binutils to 2.33.1 in Fedora 32/Rawhide and looks like I cannot recompile opensbi (there are no changes compared to F31 build).
```
BUILDSTDERR: /usr/bin/ld: /builddir/build/BUILD/opensbi-be92da280d87c38a2e0adc5d3f43bab7b5468f09/fedora-builds/uboot-sifive-fu540/build/platform/sifive/fu540/firmware/fw_payload.o(.text+0x1961): 15 bytes required for alignment to 16-byte boundary, but only 14 present
BUILDSTDERR: /usr/bin/ld: can't relax section: bad value
BUILDSTDERR: collect2: error: ld returned 1 exit status
BUILDSTDERR: make: *** [Makefile:278: /builddir/build/BUILD/opensbi-be92da280d87c38a2e0adc5d3f43bab7b5468f09/fedora-builds/uboot-sifive-fu540/build/platform/sifive/fu540/firmware/fw_payload.elf] Error 1
```
Other changes between F31 and F32 today is kernel (5.3 vs master branch) and glibc (2.30 vs master branch).

avpatel, 2019-10-22:
Strange, I had tested U-Boot-2019.10 release with OpenSBI v0.5 release.

Can you provide exact steps to reproduce this compile error ?

davidlt, 2019-10-22:
That's exactly the versions I am using. The following worked:
- QEMU virt with kernel payload;
- QEMU virt with U-Boot payload;

So only SiFive FU540 with U-Boot payload fails (tried twice on different hosts, i.e. boards).

Full build log is here (incl. commands being executed): http://fedora.riscv.rocks/kojifiles/work/tasks/5439/315439/build.log

I just tagged older binutils 2.32 to F32/Rawhide to check if this is due to changes in binutils (most likely).

avpatel, 2019-10-22:
Can you also tell us size of hifive-unleashed-a00.dtb and sifive_fu540/u-boot.bin ?

It would be even better if you can send us above binaries via google driver (or something else).

Regards,
Anup

davidlt, 2019-10-22:
Huh, 2.32 produced the same issue. You can extract those from RPMs, .e.g:
```
rpm2cpio 'http://fedora.riscv.rocks/kojifiles/packages/kernel/5.4.0/0.rc3.git2.300.1.riscv64.fc32/riscv64/kernel-core-5.4.0-0.rc3.git2.300.1.riscv64.fc32.riscv64.rpm' | cpio -dvim
```
This should be the DTB `/lib/modules/5.4.0-0.rc3.git2.300.1.riscv64.fc32.riscv64/dtb/sifive/hifive-unleashed-a00.dtb`
For the U-Boot image:
```
rpm2cpio 'http://fedora.riscv.rocks/kojifiles/packages/uboot-tools/2019.10/2.0.riscv64.fc31/noarch/uboot-images-riscv64-2019.10-2.0.riscv64.fc31.noarch.rpm' | cpio -dvim
```
That should have `/usr/share/uboot/sifive_fu540/u-boot.bin`

avpatel, 2019-10-22:
This seems to be a know issue with RISC-V binutils.

Refer, https://github.com/riscv/riscv-gnu-toolchain/issues/298

Can you try adding line ".option norelax" before line "#ifdef FW_PAYLOAD_FDT_PATH" in firmware/fw_payload.S ?

Regards,
Anup

avpatel, 2019-10-22:
Alternately, can you also try replacing ".align 4" with ".balign 16" in firmware/fw_payload.S ?

Regards,
Anup

davidlt, 2019-10-22:
`.option norelax` seems to solve the problem, but changing `.align 4` to `balign 16` does not.

avpatel, 2019-10-22:
Cool, thanks for trying my suggestion.

The issue will only manifest when we are embedding DTB in FW_PAYLOAD and DTB size is not multiple of 16 bytes.

You can send patch to OpenSBI. Please mention the GCC binutils issue in your commit description as well.

Regards,
Anup

avpatel, 2019-10-23:
I send patch to OpenSBI to fix this issue.

I was not able to reproduce this issue with Bootlin toolchain so I guess lot of people might not be seeing this issue.
(Refer, https://toolchains.bootlin.com/)

Can you provide Tested-by to my patch ?

Regards,
Anup

davidlt, 2019-10-23:
I noticed that. I just started a test build with my own patch and your patch arrived. I killed my build and pulled in your patch instead. I will send Tested-by within 30 minutes if nothing goes wrong.

One nitpick, why `.option pop` is not aligned to `.option push` in your patch?

avpatel, 2019-10-23:
You did not acknowledge that you will send patch so I thought you might be busy.

My bad, I should have waited little longer for your patch.

The ".option push" will save current options and ".options pop" will restore previously saved options. This is done same way Linux head.S as well.

Regards,
Anup

davidlt, 2019-10-23:
What I wanted to say is that indentation is different between push and pop lines.

avpatel, 2019-10-23:
Ahh, the difference in indentation is typo. I will fix it.

Regards,
Anup

avpatel, 2019-10-28:
Fix for this is now merged. Closing this issue.

Regards,
Anup

closed 2019-10-28 by avpatel