riscv-software-src/opensbi #141: rv32ima minimum possible architecture?

Ovsthus, 2019-10-23:
Hello,

I have a rv32ia core that I wish to run Linux on using opensbi, but I can't seem to
build opensbi for rv32ia because of errors like this:
opensbi/firmware/fw_base.S:180: Error: unrecognized opcode `mul a5,s8,t1'

Building it for rv32ima solves this error but now my core can't run it. 

I could add the extension but it would be good to know if I actually have to, time is a factor.

Thankful for any input

alistair23, 2019-10-23:
How are you planning on running Linux with just rv32ia? Does Linux support not having the multiply extension?

avpatel, 2019-10-24:
rv32ima with S-MODE would be minimal.

You will have to disable FPU and instruction compression in kernel and rootfs as well.

Ideally, rc32gc is recommended for 32bit and rv64gc for 64bit.

Regards,
Anup

avpatel, 2019-10-28:
I think we can close this issue. I will wait for one more day.

Regards,
Anup

alicecrawford, 2020-02-06:
I've been playing around with it lately and RISC-V Linux's minimum is rv32ia with S-mode, and the only thing keeping openSBI from running with rv32ia (at least the 0.5 version I've been poking at) is some ASM MUL instructions.  I've been using a macro in my own versions to do a soft MUL when the M extension is not available to build on rv32ia.

When I can get around to groking the style guide for openSBI I'm planning on submitting a patch, but I have other matters to attend to first.

I'm also contemplating how to use openSBI in cases without the A or supervisor-mode extensions, but that's a bit more in depth.  (I kinda feel openSBI should be available to as many extension sets as possible, as I'd want it to bootstrap even on modest microcontroller implementations)

atishp04, 2020-02-07:
I am just curious why do you need to run OpenSBI on micro controller or a platform without supervisor mode. The objective of OpenSBI is to provide SBI (supervisor binary interface) implementation. Why do you want to use OpenSBI if there is no supervisor ?

avpatel, 2020-05-31:
Closing this issue now.

Related documentation can be found in docs/platform_requirements.md

Regards,
Anup

closed 2020-05-31 by avpatel