riscv-software-src/opensbi #143: Trying to compile: Invalid -march= option: `rv32imafdc'

xefe, 2019-11-18:
Tried compiling this with riscv64-buildroot-linux-gnu and riscv64-linux toolchains (both included in bootlin tar)
` CC        lib/sbi/riscv_asm.o
Assembler messages:
Fatal error: invalid -march= option: 'rv32imafdc'
Makefile:297: recipe for target '/mnt/d/Projectos/Ubuntu-gcc/opensbi-master/build/lib/sbi/riscv_asm.o' failed
make: *** [/mnt/d/Projectos/Ubuntu-gcc/opensbi-master/build/lib/sbi/riscv_asm.o] Error 1`

Tried changing `PLATFORM_RISCV_ISA = rv$(PLATFORM_RISCV_XLEN)imafdc` on the makefile to multiple combinations of ISA extensions ("i","imac","g",etc) only "e" made a difference with "invalid ISA string" (is rv32e not supported?), also tried doing XLEN=64 and nothing worked.

I tried just making it and using PLATFORM=qemu/virt

I'm on WSL Ubuntu18.04

olofj, 2019-11-19:
bootlin tar toolchain questions should go to bootlin, they are the ones who would know how their toolchains are configured.

olofj, 2019-11-19:
Oh, the README even tells you why this doesn't work. You're trying to build a 32-bit target and their toolchain doesn't support it:

https://github.com/riscv/opensbi#required-toolchain

xefe, 2019-11-19:
I tried doing it with a 64-bit target and was the same, also the readme has a section for 32 xlen.
**I got the xPack GNU RISC-V Embedded GCC toolchain and this issue is fixed with that one**, but new issues raised. I gonna try some thing and open a new issue if unresolved.

closed 2019-11-19 by xefe