riscv-software-src/opensbi #146: Adding non-platform-specific but custom SBI?

dayeol, 2019-12-12:
Hi, 

I'm used to bbl and new to opensbi.

I found that bbl is slowing down and people are moving to opensbi.

We are research team at Berkeley building hardware enclave extension on the firmware (Keystone Enclave).

We want to move forward to opensbi so that it would easily work with more platforms.

I found that opensbi is really nice in building a platform-specific firmware.

However, we want to make our extension (having a few additional SBIs) work on multiple platforms including sifive/fu540 and qemu/virt.

So, we want the library to be included in multiple platforms.

Currently, I see that all common libs are in `lib` where all platform-specific functions should go to `platform`, and you get to choose one platform when you compile the firmware.

Do you have any plans for adding SBI extensions in the build system that works with multiple platforms? 

What is your recommendation for putting such non-standard extension in opensbi?

Thank you!

dayeol, 2019-12-13:
I actually figured out after trying several options.

I think the best way still is to add a directory under "platform" and have a common source files in the objects.mk under each platform sub-directory (e.g., `keystone/qemu/virt`), as the additional extension is for specific set of platforms.

Can I add a platform named `keystone`? We will make the code comply with your contribution guideline.

avpatel, 2019-12-13:
Yes, you can add a duplicate platform but this is not the best way to do it.

In SBI v0.2, we have reserved range for "Experimental Extensions" and "Platform/Vendor Extensions".

A "Platform/Vendor SBI Extension" is strictly vendor/platform specific and cannot be shared across platforms whereas "Experimental SBI Extension" is for work-in-progress SBI extensions (such as Keystone Enclave SBI calls) and can be shared across platforms.

The SBI v0.2 patches have just landed in OpenSBI repo and SBI v0.2 replacement extensions are still under review.

Currently, we don't have a mechanism to register/unregister "Experimental SBI Extensions" at runtime so we need that first for your "Keystone Enclave SBI calls". I will add this mechanism for you so stay tuned.

Once we have above mentioned mechanism, you can add "Keystone Enclave" implementation under lib/utils/experimental of your OpenSBI fork and you can invoke "Keystone Enclave" init() function from platform final_init() to play with it.

Eventually, we will have to define "Keystone Enclave" in SBI spec and we can merge your "Keystone Enclave" in OpenSBI repo only after it is defined in a SBI draft/ratified spec.

Does it sound good to you ?

Regards,
Anup

dayeol, 2019-12-13:
Thanks for the explanation. It sounds perfect! I will stay tuned for the update.

Only one question: does "register/unregister experimental SBI extension at runtime" mean that we make it loadable and dynamically assign SBI numbers? Or does that simply mean I need to invoke keystone "init()" in the platform's final_init()?

atishp04, 2019-12-15:
I think it's the latter. I have few other comments.

1. The keystone enclave extension must follow SBI v0.2 calling convention.
2. The custom SBI extension documentation should be made public in order to get feedback from the community. This will also help in merging the specification to SBI spec later.

avpatel, 2020-01-16:
I have sent-out a patch for this to OpenSBI mailing list.

To join, OpenSBI mailing list visit:
http://lists.infradead.org/mailman/listinfo/opensbi

Regards,
Anup

avpatel, 2020-01-23:
Required patches are reviewed and merged.

Regards,
Anup

closed 2020-01-23 by avpatel