riscv-software-src/opensbi #147: Dynamic PMP Synchronization

dayeol, 2019-12-13:
Hi,

I have another question regarding PMP.
Currently, I see that PMP is initialized at `sbi_hart_init()`, which is one-time called per-hart at boot.

Keystone Enclave uses PMP for dynamic memory isolation, and all of the PMP entries in all cores must be synchronized when an enclave is created/destroyed.

In bbl version of keystone, we have done it by adding a custom IPl, where one core initiates the IPI to update PMP entries and then waits until all the other cores finishes updating PMP.

I see there is no way for now to cleanly add a custom IPI without changing the core functions like `sbi_ipi_process()`.
Do you have any suggestions on what would be the best way to add this?

avpatel, 2019-12-13:
For IPI, we will provide register/unregister mechanism just like experimental SBI extensions.

Regards,
Anup

dayeol, 2019-12-14:
Wonderful. Thanks!

avpatel, 2020-01-16:
I have sent-out a patch series for this to OpenSBI mailing list.

To join, OpenSBI mailing list visit:
http://lists.infradead.org/mailman/listinfo/opensbi

Regards,
Anup

avpatel, 2020-01-23:
Patches are reviewed and merged. Please try at your end.

Regards,
Anup

closed 2020-01-23 by avpatel