riscv-software-src/opensbi #148: The right way for instruction emulation in OpenSBI ?

Ruinland-ChuanTzu-Tsai, 2019-12-23:
Hi,
I'm interested in the possibility to port OpenSBI for [Linux-on-VexRiscv](https://github.com/litex-hub/linux-on-litex-vexriscv) , which uses its own SBI implementation.
Yet the issue is that A extension is not fully support in VexRiscv and the unsupported ones are [emulated in its SBI](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/emulator/main.c#L427-L463) .
And if memory serves me right, in RISC-V's unprivileged specification, this kind of simulation is mentioned as well.

I know there's a [illegal instruction handling part](https://github.com/riscv/opensbi/blob/master/lib/sbi/sbi_illegal_insn.c#L136) yet it seems to be too generic.
I'd like to know is there a platform specific way to do this ?

avpatel, 2019-12-24:
I think it's more complicated than simply adding AMO instruction emulation because OpenSBI uses AMO instructions for itself.

Please Refer,
lib/sbi/sbi_init.c
lib/sbi/sbi_tlb.c
lib/sbi/sbi_ipi.c
lib/sbi/riscv_atomc.c
firmware/fw_base.S

The RISC-V extensions mandatory for OpenSBI are I, M, and A.

We would certainly like to add support for VexRiscv but I would insist that VexRiscv implement RISC-V A extension completely rather than emulating AMO in OpenSBI.

Ruinland-ChuanTzu-Tsai, 2019-12-25:
> The RISC-V extensions mandatory for OpenSBI are I, M, and A.
> We would certainly like to add support for VexRiscv but I would insist that VexRiscv implement RISC-V A extension completely rather than emulating AMO in OpenSBI.

Point taken.

That being said, though could be stretched too far, I'm still curious about about the "instruction emulation" scenario. It is possible that people may try to emulate a subset of extension, not necessarily A, in M-mode firmware.
Is there recommended approach in OpenSBI for this kind of needs ?

avpatel, 2019-12-25:
We don't have a defined policy regarding emulation of other RISCV extensions in OpenSBI.

Till now, we have tried to keep instruction emulation at minimum in OpenSBI. That's why we have not added F and D extension emulation in OpenSBI because we can always have FP disabled in Linux and soft-float Linux rootfs.

I think adding emulation of any RISC-V extension in OpenSBI should only be last resort because trap-n-emulate not only adds lot of code but it is slow as well.

All this needs to be discussed in larger forum because OpenSBI is now used by lot of platforms. Probably you can start an email thread regarding this on OpenSBI mailing list.

Regards,
Anup

avpatel, 2020-01-23:
No activity on this issue. Please re-open this issue if required.

Regards,
Anup

closed 2020-01-23 by avpatel

gsomlo, 2021-11-09:
@avpatel -- based on your comments earlier in this thread, do you think that maybe this https://github.com/riscv-software-src/opensbi/blame/master/docs/platform_requirements.md#L40 should be reworded so as to avoid suggesting that OpenSBI might in fact currently support emulation of (at least) F and D? Thanks!