riscv-software-src/opensbi #150: Can opensbi support smp directly? without uboot

kiutao, 2020-01-17:
I try to boot linux  using opensbi directly. 
one core: success
8cores:
[ 0.000992] smp: Bringing up secondary CPUs …
[ 1.057750] CPU1: failed to come online
.....
only one core can work..

kiutao, 2020-01-17:
solved.

closed 2020-01-17 by kiutao

MarcKarasek, 2020-06-11:
How did you solve this?  I am seeing the same thing under qemu.  I recently upgraded to qemu 6 from 5.

avpatel, 2020-06-11:
For OpenSBI v0.7 (or higher), use Linux-5.7 (or higher).

The OpenSBI v0.7 implement SBI HSM extension which brings ordered booting of HARTs on SMP host.

If you want to use older Linux with OpenSBI v0.7 (or higher) then you can back-port Linux SBI v0.2 support patches.

I hope above info helps.

Thanks,
Anup