riscv-software-src/opensbi #152: How to payload elf for debug using?

sequencer, 2020-01-22:
I'm trying to debug my own program boatloaded by OpenSBI with `FW_PAYLOAD`.
currently binary can be payloaded by OpenSBI with `FW_PAYLOAD`.
If I wanna connect my FPGA to debug this program with gdb elf, how can I do?

alistair23, 2020-01-23:
You will need to run OpenOCD and then connect GDB to that. You will need to follow the documentation provided by the CPU core.

closed 2020-01-23 by alistair23

sequencer, 2020-01-23:
I successfully loaded opensbi elf with gdb and openocd.
Question is if opensbi only load bin file. Metadata like sources information can not be loaded to opensbi elf that we cannot debug payload application.

avpatel, 2020-01-23:
Metadata info like debug symbols are not useful for any Bootloader or Firmware so OpenSBI only includes flat image as payload.

For debugging using GDB/OpenOCD, users have to load debug info from ELF separate.

Above facts are true for all architectures.

sequencer, 2020-01-23:
So I should use `FW_JUMP` firstly. Break at jump address. Then load my debug elf to jump address and continue?

alistair23, 2020-01-23:
You can use the `FW_PAYLOAD` and set a break at the jump address

sequencer, 2020-01-23:
Yes. I think I need to load payload debug metadata from elf.

asyarifstudio, 2023-03-06:
Hi, I'm, trying to do the same, basically debugging the linux kernel which is the payload of the openSBI.

I can debug the openSBI itself with this VS Code configuration
```
{
            "name": "CVA6 - SM Debugger",
            "type": "cppdbg",
            "MIMode": "gdb",
            "request": "launch",
            "miDebuggerPath": "${workspaceRoot}/riscv64/bin/riscv64-unknown-elf-gdb",
            "program": "${workspaceRoot}/build/sm.build/platform/fpga/ariane/firmware/fw_payload.elf",
            "miDebuggerServerAddress": "localhost:3333",
            "cwd": "${workspaceRoot}/sm"
 }
```
but what configuration should I use? I tried to set the `program` to `${workspaceRoot}/build/linux.build/vmlinux` and change the `cwd` to `${workspaceRoot}/linux` but I still not able to debug the linux kernel.

do you have any sample or tutorial I can follow?