riscv-software-src/opensbi #154: Can't clear supervisor timer interrupt

josecm, 2020-01-28:
I'm running a simple bare-metal application over sbi on qemu virt. I use the sbi_set_timer function to set a tick at a given interval. It seems to work fine. However, when I try to disable the timer interrupts by calling `sbi_set_timer((uint64_t)-1)` it seems to have the inverse effect: the interrupt is triggered consecutively ad infinitum.

I've set a breakpoint at `sbi_timer_event_start` and confirmed the `nex_event` argument is indeed ((uint64_t)-1).

Am I doing something wrong? Or maybe this is a QEMU issue?

avpatel, 2020-02-02:
I think the best thing to do is what Linux RISC-V timer driver does:
1. Upon new timer event, enable timer interrupt bit is SIE CSR and do SBI set timer call
2. Upon every timer interrupt, disable timer interrupt bit in SIE CSR

Apart from above to force clear timer, we just need to clear timer interrupt enable bit in SIE CSR. In this case, setting next timer event to (u64)-1 is good to do but not required.

Can you try above ?

Regards,
Anup

josecm, 2020-02-03:
I forgot to mention that clearing the STIE in SIE CSR it is exactly what I'm doing to achieve this. So, that works.

Thanks,
José

avpatel, 2020-10-21:
No activity on this issue hence closing it.

closed 2020-10-21 by avpatel

blitz, 2020-12-28:
For future people coming here:

One pitfall on 32-bit RISC-V is that the timer register and the time argument to the `set_timer` SBI call are **always 64-bit**. If you truncate to 32-bit, you will program timestamps in the past and get a new interrupt right away.

This is not a bug in OpenSBI, but something that needs to be watched out for nonetheless.