riscv-software-src/opensbi #157: delegate_traps requires medeleg[0] to be writable

pdonahue-ventana, 2020-02-12:
In section 3.1.8, the privileged architecture spec says:
> An implementation can choose to subset the delegatable traps, with the supported delegatable bits found by writing one to every bit location, then reading back the value in medeleg or mideleg to see which bit positions hold a one.

Presumably, an implementation can hard-wire medeleg to 0.

The [delegate_traps](https://github.com/riscv/opensbi/blob/master/lib/sbi/sbi_hart.c#L72) function assumes that various traps are delegatable and are not hard-wired to 0 (at minimum: misaligned fetches, breakpoints, and user ecalls).  If any of the bits that it wants to delegate are hard-wired to 0 then it will return a failure which prevents that hart from booting.

I can understand that there are some minimum requirements for OpenSBI to run (such as misa being implemented and possibly breakpoints and user ecalls being delegatable, though I can't find these requirements documented anywhere).  However...

The privileged spec says:
> For exceptions that cannot occur in less privileged modes, the corresponding medeleg bits should be hardwired to zero.

and the unprivileged spec says:
> With the addition of the C extension, no instructions can raise instruction-address-misaligned exceptions.

So if misa.C is hard-wired to 1 then the spec says that medeleg[0] should be hard-wired to 0 but OpenSBI requires that bit to be writable.  To resolve the conflict, I think that OpenSBI should only require medeleg[0] to be writable in implementations where misa.C can be 0.

This was found by inspection so it's possible that I'm missing something.

avpatel, 2020-02-12:
We had added CSR_MIDELEG and CSR_MEDELEG sanity checks in early days of OpenSBI for bring-up on SiFive Unleashed and Kendryte K210.

Considering the fact that some of the bits in CSR_MIDELEG and CSR_MEDELEG can be hardwired to fixed values, it does not make sense to keep the CSR_MIDELEG and CSR_MEDELEG sanity checks.

I suggest we drop these sanity checks and instead add boot-time prints showing value programmed in CSR_MIDELEG and CSR_MEDELEG.

Suggestions ??

avpatel, 2020-02-18:
I have sent patches to remove checks which were enforcing certain MIDELEG and MEDELEG bits to be writable. For now, we just print MIDELEG and MEDELEG at boot time so that users know final configured values of these CSRs.

The only minimum requirement for OpenSBI is RV32GC or RV64GC. If there is any other implicit requirement (such as writeablity of certain bits in CSRs) then we should certainly try to remove such implicit requirements.

Like you mentioned, we certainly need to have a minimum HW requirements documented somewhere in OpenSBI documentation.

pdonahue-ventana, 2020-02-18:
This sounds great.  Thanks.

alistair23, 2020-04-28:
Closing this as it seems fixed.

closed 2020-04-28 by alistair23