riscv-software-src/opensbi #158: Booting opensbi+linux under qemu ..network tap issue.

MarcKarasek, 2020-02-25:
I have used busybear ( https://github.com/michaeljclark/busybear-linux ) to help me boot opensbi+linux payload

I have setup tap0 on my ubuntu18 (VM) and have a br0 bridge configured.

The Host interface is a Virtualbox Host-Only Interface. I just want to talk with the riscv machine from the HOST..

I am booting riscv on qemu-system-riscv64:

qemu-system-riscv64 -M virt -m 1024M -nographic
-kernel images/opensbi_linux.elf
-drive file=images/ramdisk,format=raw,id=hd0
-device virtio-blk-device,drive=hd0
-append "root=/dev/vda rw console=ttyS0"
-netdev type=tap,ifname=tap0,script=ramdisk/scripts/ifup.sh,downscript=ramdisk/scripts/ifdown.sh,id=net0
-device virtio-net-device,netdev=net0

I have configured the eth0 on the image as both static ip and dchp.

When dhcp I see the request on the Host and it gets an ipaddr during boot. But it never shows up in ifconfig. And I cannot ping the HOST.

If i configure it as static ifconfig shows the address but no route to the outside.

Both are going through the same tap0 interface above.
bridge name bridge id STP enabled interfaces
br0 8000.080027ea438c no enp0s8
tap0

I am at a loss now as the bridge seems to be working at least for dhcp....

alistair23, 2020-02-25:
This doesn't seem to be an OpenSBI issue. It sounds like a configuration issue. [QEMU TAP networking](https://wiki.qemu.org/Documentation/Networking) can be difficult to configure. I would recommend starting with a tool that does it for you and then make any changes you need from there. OpenEmbedded will automatically configure TAP devices for you.

closed 2020-02-25 by alistair23

MarcKarasek, 2020-02-25:
Can't really do this..  AS this is the baseline for a real SOC environment.. 
The TAP device is configured and is working at least for dhcp.  You see a lease given in the riscv VM.  But never makes it to the interface (eth0)..