riscv-software-src/opensbi #160: Unable to find the preferred boot HART id

musamaanjum, 2020-03-11:
I've run FSBL on all 5 cores of the HiFive Unleased board and then launching the openSBI on all 5 cores by building the openSBI(078686d7) by: 
`make PLATFORM=sifive/fu540 FW_PAYLOAD_PATH=u-boot-gitlab/u-boot.bin`
OR
`make PLATFORM=sifive/fu540 FW_PAYLOAD_PATH=u-boot-gitlab/u-boot.bin FU540_ENABLED_HART_MASK=0x02`
In both cases, the `_boot_status` is never being set to `BOOT_STATUS_BOOT_HART_DONE` and all cores keeps on waiting for the _boot_status to be updated. What is wrong here?

The workaround is to specify the boot core id in `fw_boot_hart` function avoiding `_try_lottery` function which don't do any good.

avpatel, 2020-03-11:
Don't use FU540_ENABLED_HART_MASK, it's not required anymore.

Other than that we don't see any issues. Please share detailed bootlog.

Regards,
Anup

musamaanjum, 2020-03-11:
Without using `FU540_ENABLED_HART_MASK`, I still face the issue. How can I collect the bootlog?

nmeum, 2020-03-27:
I believe I am facing the same issue, even with current git head (where `FU540_ENABLED_HART_MASK` is removed) I am unable to boot the OpenSBI sample payload with `fw_payload.elf` for the `sifive/fu540` platform with QEMU 4.2 using the `sifive_u` machine. As described in this issue, OpenSBI seems to hang in `_wait_for_boot_hart`.

avpatel, 2020-03-27:
Please try OpenSBI v0.6 release.

To use latest OpenSBI, you will need SBI HSM support in Linux kernel to bring up CPUs.

For details refer, Atish's talk on ordered booting in FOSDEM 2020.

My apology for delayed response, I lost track of this GitHub issue.

Regards,
Anup

nmeum, 2020-03-27:
> To use latest OpenSBI, you will need SBI HSM support in Linux kernel to bring up CPUs.

I am not even using the Linux kernel, I am just trying to use the OpenSBI sample payload.

I build OpenSBI as follows:

    make CROSS_COMPILE=riscv64-unknown-elf- PLATFORM=sifive/fu540

And started the payload in QEMU 4.2 using:

    qemu-system-riscv64 -M sifive_u -nographic -kernel build/platform/sifive/fu540/firmware/fw_payload.elf -s

Attaching GDB yields the following output:

```
_wait_for_boot_hart () at /home/nmeum/src/opensbi/firmware/fw_base.S:336
336             li      t0, BOOT_STATUS_BOOT_HART_DONE
```

As outlined in the issue description, `_wait_for_boot_hart` never returns as it seems that the `_boot_status` is never set to `BOOT_STATUS_BOOT_HART_DONE`. I can also reproduce this with OpenSBI `v0.6` by the way. Let me know if I am doing something obvious wrong here. The `qemu/virt` platform does work for me though.

alistair23, 2020-03-27:
What version of QEMU are you using?

nmeum, 2020-03-27:
QEMU 4.2.0, let me know if you are not capable of reproducing this issue.

alistair23, 2020-04-02:
I just tried and I don't see any issues. I can boot the OpenSBI payload with both `-kernel` and `-bios`.

It seems like a problem with your setup. You might want to try providing QEMU with more memory, as that might be the problem.

closed 2020-04-02 by alistair23