riscv-software-src/opensbi #163: why csrr MTVEC

Dolu1990, 2020-04-27:
Hi,

Basicaly i was trying to strip down the CSR requirements of openSBI to make a softcore smaller/faster, and hit the following : 

https://github.com/riscv/opensbi/blob/e5a7f556ce22984128bed0a336551a968c818450/firmware/fw_base.S#L404

Is there some CPU in the wildness which do not update MTVEC properly ?
I'm asking to understand why that test was implemented.

Regards
Charles

avpatel, 2020-04-27:
This was added in early days of OpenSBI (15 months back) when we were bringing up on SiFive Unleashed board.

We certainly don't need this check and it can be removed.

Please send a patch to OpenSBI mailing list.

Regards,
Anup

jrtc27, 2020-05-02:
The check is there because chips are free to enforce whatever arbitrary legalisation of mtvec they like (e.g. not supporting the vectoring interrupt mode that OpenSBI expects), and it is much better to fail fast rather than to die later in boot because it turns out your trap handler wasn't installed properly.

Not supporting reading mtvec is also in direct violation of the privileged specification:

> The mtvec register must always be implemented, but can contain a hardwired read-only value.

Moreover, if you are running OpenSBI, you presumably have S-mode (otherwise why on earth do you care about the _Supervisor_ Binary Interface?), and therefore also an stvec. Illegal instructions are not delegated by OpenSBI, as it needs to emulate some of them. But, if it decides an instruction is truly illegal, it needs to pretend that the exception _was_ delegated, and to do so it needs to be able to read stvec, put it in mepc and mret (along with setting up the rest of the CSR state to look like it was a trap to S-mode). Therefore, if stvec needs to be readable for that to work, it seems very wrong to not have mtvec also readable, even ignoring the violation of the privileged spec.

avpatel, 2020-05-02:
Apart from this check, we have unpriv access in OpenSBI where we save/restore MTVEC to capture traps while doing unpriv access.

The MTVEC CSR should certainly be readable to comply with the spec as well.

For now, we have removed the MTVEC check because it was without any explanation and it was in wrong place as well.

Ideally, we should check MTVEC once after it is first written on boot HART in OpenSBI.

Regards,
Anup

Dolu1990, 2020-05-02:
About what the RISC-V spec say / requirements, i think that's a question of perspective. Let's me expose my point of view : 

The RISC-V previlegied specify all sorts of things, more or less useless, and seem to be to be designed with ASIC hardware cost in mind (not FPGA).

For instance, it ask to implement mcycle/minstret (mendatory, right ?), which are 64 bits counters.
the corresponding cost on a Artix7 FPGA is about 2*(0.75 LUT + 64 FF) = 96 LUT + 128 FF

To put that in perspective, a Linux capable SMP core is about 2700 LUT+ 2200 FF (https://groups.google.com/d/msg/linux-litex/f0NaNtXYCds/_sx6wGwvBgAJ) => implementing mcycle/minstret would cost about  3.5% of total LUT + 5.6 % of total FF, which is not acceptable in a FPGA design, unless you realy need those two features. Also, bigger design also mean slower frequency in general. And the good thing is that if you need those feature, you just need to regenerate the FPGA bitstream and flash it (hardware become like software).

Overall, there is the current CSR features set which can boot openSBI / Linux :
https://github.com/SpinalHDL/VexRiscv/blob/smp/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L87
(Note, satpAccess is marked as NONE, but is implemented somewere else)

So to resume, the less used, the better for FPGA softcore ^^

> to do so it needs to be able to read stvec

My point was about mtvec, stvec as read/write is fine to me.

jrtc27, 2020-05-02:
> About what the RISC-V spec say / requirements, i think that's a question of perspective. Let's me expose my point of view :

It's not a question of perspective. The specification clearly states when CSRs must be implemented, and whether they are RO or RW. As far as I know there are no WO CSRs in the entire privileged specification. If you decide to violate what the specification says, software may still work, but you can no longer claim that your core implements the privileged specification; instead, you have made a core that (hopefully) implements the RV32I base ISA, and a non-standard privileged extension that happens to be somewhat compatible with the RISC-V one. I didn't write the spec, I don't make the rules, I'm just telling you how things are, and I'm not going to repeat myself yet again.

> The RISC-V previlegied specify all sorts of things, more or less useless, and seem to be to be designed with ASIC hardware cost in mind (not FPGA).
>
> For instance, it ask to implement mcycle/minstret [...]

To call having mcycle or minstret "useless" is ludicrous. *You* may not have a use for them, but if you ever want to do any kind of performance analysis of your core you will soon realise that they are vital as giving you one of the most fundamental statistics people care about, CPI. Many people will gather such statistics on their FPGA soft cores without having any interest in ASICs.

> Overall, there is the current CSR features set which can boot openSBI / Linux :
> https://github.com/SpinalHDL/VexRiscv/blob/smp/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L87
> (Note, satpAccess is [...]

I don't doubt that but there are limits.

> > to do so it needs to be able to read stvec
> 
> My point was about mtvec, stvec as read/write is fine to me.

As Anup has mentioned, lib/sbi/sbi_unpriv.c already requires the ability to read mtvec in order to temporarily change where mtvec points when making an unprivileged access for which trapping should be ignored. I see zero reason to make a fuss about exposing a single CSR as readable, especially when stvec is already readable and the uniformity of the CSR encodings means it should come with barely any cost.

Dolu1990, 2020-05-02:
> It's not a question of perspective. The specification clearly states when CSRs must be implemented, and whether they are RO or RW. 
> I don't make the rules, I'm just telling you how things are.

Doesn't mean openSBI has to stick to it, instead it can limit itself to a given subset.

> To call having mcycle or minstret "useless" is ludicrous.

I should have been more eplicite, as this is relative to who the question is asked and the application. 
Some people may try to fit as many core as they can in a given FPGA, those people will try to remove all superfluous features to fit the exact requirements of their application. They will try to avoid feature creep.
Some people will want to benchmark the design, and for those poeple mcycle/minstret is needed.

> Many people will gather such statistics on their FPGA soft cores without having any interest in ASICs.

And many others will put FPGA into products, where they will try to reduce cost as much as possible by using the smallest FPGA possible.

My wish is to not have the needs of some people conflicting with the needs of others.

> As Anup has mentioned, lib/sbi/sbi_unpriv.c already requires the ability to read mtvec

A simple workaround to me seem to store the mtvec value in ram aswell than in the CSR. Same for medeleg / mideleg.

> I see zero reason to make a fuss about exposing a single CSR as readable

People trying to save Area/Money

Dolu1990, 2020-05-02:
> You may not have a use for them, but if you ever want to do any kind of performance analysis

You can update FPGA design by flashing them, everybody can then have their own flavor, then everybody is happy ^^

avpatel, 2020-05-02:
We cannot avoid reading MTVEC CSR in OpenSBI because when OpenSBI is used as library (e.g. EDK2 RISC-V), the original exception vector will be with external firmware (not part of OpenSBI sources) and to temporarily change exception vector OpenSBI will have to save/restore MTVEC.

Currently, OpenSBI easily runs on any rv64g, rv64gc, rv32g, and rv32gc. This is same as Linux RISC-V requirements.

I certainly appreciate efforts to reduce area of RISCV implementation but this should not be done by enforcing additional restrictions on-top-of RISC-V spec.

Regards,
Anup

Dolu1990, 2020-05-02:
> We cannot avoid reading MTVEC CSR in OpenSBI because when OpenSBI is used as library (e.g. EDK2 RISC-V), the original exception vector will be with external firmware (not part of OpenSBI sources) and to temporarily change exception vector OpenSBI will have to save/restore MTVEC.

Hoo right.

> Currently, OpenSBI easily runs on any rv64g, rv64gc, rv32g, and rv32gc. This is same as Linux RISC-V requirements.

I'm not exactly sure how much rv32g is a requirements for linux, currently it can run with only rv32ima. Also i think that enforcing rv32g as minimal feature set for linux would be a mistake, as implementing the floating point 32 bits + 64 bits would probably endup into to much overhead for many applications on FPGA, and is not usefull for quite many applications.

All good for me ^^
We can close the issue, unless there is something pending ?

avpatel, 2020-05-02:
Based on current state of OpenSBI, rv32ima or rv64ima is totally fine. We are not enforcing f and d extension for OpenSBI. I guess we should document required RISC-V isa features by OpenSBI somewhere.

OpenSBI will certainly work fine for your rv32ima system.

Regards,
Anup

Dolu1990, 2020-05-02:
> Based on current state of OpenSBI, rv32ima or rv64ima is totally fine. We are not enforcing f and d extension for OpenSBI. I guess we should document required RISC-V isa features by OpenSBI somewhere.

Ahh sorry, maybe i wasn't clear, I was refering to the linux kernel minimal requirements, and by only  rv32ima, i meant, that the linux kernel is fine with rv32ima without float/double.

> I guess we should document required RISC-V isa features by OpenSBI somewhere.
So personnaly, before switching to opensbi, i had my own machine mode SBI implementation.  (https://github.com/SpinalHDL/VexRiscv/tree/master/src/main/c/emulator/src), but i'm quite happy to migrate to openSBI, and the only additional feature i had to enable was MTVEC read, mideleg read and medeleg read. Else, all worked out of the box :)

> OpenSBI will certainly work fine for your rv32ima system.
Yes it work well, through openSBI it can boot linux on a quad-core, on a 129$ FPGA dev kit :D

Regards
Charles

closed 2020-05-02 by Dolu1990

avpatel, 2020-05-02:
For Linux, rv32gc or rv64gc is default currently. We have to explicitly disable FP if we don't want f and d extension. Also, we will have to explicitly build soft-fp rootfs.

AFIK, all major distros have chosen to only support rv32gc and rv64gc. No distros will support soft-FP rootfs by default.

In RISC-V world, if users want things to work out of the box then rv32gc or rv64gc is right way to go.

Regards,
Anup

Dolu1990, 2020-05-02:
> For Linux, rv32gc or rv64gc is default currently. We have to explicitly disable FP if we don't want f and d extension. Also, we will have to explicitly build soft-fp rootfs.

That's many thing i'm not realy aware of, as i'm more in the hardware side of things ^^

> AFIK, all major distros have chosen to only support rv32gc and rv64gc. No distros will support soft-FP rootfs by default.

So maybe here, there should be a distinctions between (i might be wrong) : 
- Desktop distro, with package manager / updater (ex Debian, ...)
- Embedded distro, which are made to go on embedded device, hidden from the users, and where package manager is only when you compile the distro (ex : Buildroot, openwrt), which are more on the industrial side of things. Basicaly, on such distro, nobody has to handle online repositories of compiled packages.

(I'm targeting embedded distro)

avpatel, 2020-05-02:
By distros, I mean Fedora, Debian, SUSE, Yocto, Buildroot, etc. We might see RHEL and Ubuntu doing RISC-V efforts soon as well.

For embedded, Busybox is easy to use for all purposes (including soft-fp rootfs).

Regards,
Anup

davidlt, 2020-05-02:
For Linux distributions the decision is somewhat easy as RV64GC is required by "RISC-V UNIX-Class Platform Specification" https://github.com/riscv/riscv-platform-specs/blob/master/riscv-unix.adoc

It might in the future define different profiles there full GC is not required (e.g. embedded targets).