riscv-software-src/opensbi #166: master branch can't build, v0.7 is OK

gongbo2018, 2020-05-29:
```
wgb@wgb-virtual-machine:~/riscvSpace/opensbi$ export CROSS_COMPILE=riscv64-linux-
wgb@wgb-virtual-machine:~/riscvSpace/opensbi$ make PLATFORM=qemu/virt FW_PAYLOAD_PATH=../u-boot/u-boot.bin
 CC-DEP    lib/utils/fdt/fdt_fixup.dep
 CC-DEP    lib/utils/fdt/fdt_helper.dep
 CC-DEP    lib/utils/ipi/fdt_ipi_clint.dep
 CC-DEP    lib/utils/ipi/fdt_ipi.dep
 CC-DEP    lib/utils/irqchip/plic.dep
 CC-DEP    lib/utils/irqchip/fdt_irqchip_plic.dep
 CC-DEP    lib/utils/irqchip/fdt_irqchip.dep
 CC-DEP    lib/utils/libfdt/fdt_wip.dep
 CC-DEP    lib/utils/libfdt/fdt_sw.dep
 CC-DEP    lib/utils/libfdt/fdt_strerror.dep
 CC-DEP    lib/utils/libfdt/fdt_rw.dep
 CC-DEP    lib/utils/libfdt/fdt_ro.dep
 CC-DEP    lib/utils/libfdt/fdt_empty_tree.dep
 CC-DEP    lib/utils/libfdt/fdt_addresses.dep
 CC-DEP    lib/utils/libfdt/fdt.dep
 CC-DEP    lib/utils/reset/fdt_reset_sifive.dep
 CC-DEP    lib/utils/reset/fdt_reset_htif.dep
 CC-DEP    lib/utils/reset/fdt_reset.dep
 CC-DEP    lib/utils/serial/uart8250.dep
 CC-DEP    lib/utils/serial/sifive-uart.dep
 CC-DEP    lib/utils/serial/fdt_serial_uart8250.dep
 CC-DEP    lib/utils/serial/fdt_serial_sifive.dep
 CC-DEP    lib/utils/serial/fdt_serial_htif.dep
 CC-DEP    lib/utils/serial/fdt_serial.dep
 CC-DEP    lib/utils/sys/sifive_test.dep
 CC-DEP    lib/utils/sys/htif.dep
 CC-DEP    lib/utils/sys/clint.dep
 CC-DEP    lib/utils/timer/fdt_timer_clint.dep
 CC-DEP    lib/utils/timer/fdt_timer.dep
 AS-DEP    lib/sbi/sbi_expected_trap.dep
 CC-DEP    lib/sbi/sbi_unpriv.dep
 CC-DEP    lib/sbi/sbi_trap.dep
 CC-DEP    lib/sbi/sbi_tlb.dep
 CC-DEP    lib/sbi/sbi_timer.dep
 CC-DEP    lib/sbi/sbi_system.dep
 CC-DEP    lib/sbi/sbi_string.dep
 CC-DEP    lib/sbi/sbi_scratch.dep
 CC-DEP    lib/sbi/sbi_platform.dep
 CC-DEP    lib/sbi/sbi_misaligned_ldst.dep
 CC-DEP    lib/sbi/sbi_ipi.dep
 CC-DEP    lib/sbi/sbi_init.dep
 CC-DEP    lib/sbi/sbi_illegal_insn.dep
 CC-DEP    lib/sbi/sbi_hsm.dep
 AS-DEP    lib/sbi/sbi_hfence.dep
 CC-DEP    lib/sbi/sbi_math.dep
 CC-DEP    lib/sbi/sbi_hart.dep
 CC-DEP    lib/sbi/sbi_fifo.dep
 CC-DEP    lib/sbi/sbi_emulate_csr.dep
 CC-DEP    lib/sbi/sbi_ecall_vendor.dep
 CC-DEP    lib/sbi/sbi_ecall_replace.dep
 CC-DEP    lib/sbi/sbi_ecall_legacy.dep
 CC-DEP    lib/sbi/sbi_ecall_hsm.dep
 CC-DEP    lib/sbi/sbi_ecall_base.dep
 CC-DEP    lib/sbi/sbi_ecall.dep
 CC-DEP    lib/sbi/sbi_console.dep
 CC-DEP    lib/sbi/sbi_bitops.dep
 CC-DEP    lib/sbi/sbi_bitmap.dep
 CC-DEP    lib/sbi/riscv_locks.dep
 AS-DEP    lib/sbi/riscv_hardfp.dep
 CC-DEP    lib/sbi/riscv_atomic.dep
 CC-DEP    lib/sbi/riscv_asm.dep
Makefile:117: /home/wgb/riscvSpace/opensbi/platform/qemu/virt/config.mk: 没有那个文件或目录
make: *** 没有规则可制作目标“/home/wgb/riscvSpace/opensbi/platform/qemu/virt/config.mk”。 停止。

```

avpatel, 2020-05-29:
qemu/virt platform is replaced by generic platform in latest master.

Use "make PLATFORM=generic" to build OpenSBI for qemu/virt.

You can also refer latest docs/platform/qemu_virt.md

Regards,
Anup

avpatel, 2020-05-31:
Closing this issue since it works for most of us.

Regards,
Anup

closed 2020-05-31 by avpatel