riscv-software-src/opensbi #173: init_count is uninitialized

aharris-ventana, 2020-07-01:
While running opensbi on our simulator with valgrind, I've found several points where code in opensbi performs operations on uninitialized memory.

I don't have the time to fully root-cause these issues, but hopefully I'll provide enough information to enable a quick debug session.

This code increments an uninitialized memory location in hart0's scratch area:
```
sbi_init.c:215

init_count = sbi_scratch_offset_ptr(scratch, init_count_offset);
(*init_count)++;   <-- *init_count is not initialized. 

```
A list of accesses to addresses to this area of memory (from our internal traces):
```
    [0x80014058] <- 0x0000000000000002    ; <stack[47]>
    [0x80014050] -> 0x0000000000000000    ; <stack[38]>  <-- no previous write
    [0x80014050] <- 0x0000000000000001    ; <stack[38]>
    [0x80014058] -> 0x0000000000000002    ; <stack[43]>
    [0x80014058] <- 0x0000000000000003    ; <stack[43]>
```

Our build was a 'generic' flavor from SHA: 89ba63493c4ce98861144180e408d6db391fd1c3

avpatel, 2020-07-06:
The init_count is initialized to zero from sbi_scratch_alloc_offset() on boot HART.

Can you double check ??

Regards,
Anup

avpatel, 2020-10-21:
I assume this is resolved. If so please close this issue.

closed 2020-10-23 by avpatel