riscv-software-src/opensbi #174: Building for no payload fail

montjoie, 2020-07-02:
I tried what said the documentation and it fail to build:
First the documentation forgot to say to use configure, so I did:
./configure --host=riscv64-buildroot-linux-gnu

Then I did
make PLATFORM=generic
riscv64-buildroot-linux-gnu-ranlib libdummy_payload.a
riscv64-buildroot-linux-gnu-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static    -fno-stack-protector -o dummy_payload dummy_payload.o -L.  -ldummy_payload -lgcc -Wl,--defsym=MEM_START=0x80000000,-T,./dummy_payload/dummy_payload.lds
/srv/data/clabbe/toolchains/riscv64--glibc--bleeding-edge-2020.02-2/bin/../lib/gcc/riscv64-buildroot-linux-gnu/9.3.0/../../../../riscv64-buildroot-linux-gnu/bin/ld: cannot open output file dummy_payload: est un dossier
collect2: error: ld a retourné le statut de sortie 1
Makefile:332: recipe for target 'dummy_payload' failed
make: *** [dummy_payload] Error 1

avpatel, 2020-07-06:
There is no dummy_payload in OpenSBI makefiles.

Can you double check your build-system ??

Regards,
Anup

avpatel, 2020-08-07:
I am assuming this is resolved at your end hence closing issue.

Regards,
Anup

closed 2020-08-07 by avpatel