riscv-software-src/opensbi #176: what is the main function of "sbi_hart_set_trap_info(scratch, trap);" and "sbi_hart_set_trap_info(scratch, NULL);embrace code?

13824125580, 2020-08-11:
Hi folks:
 
    in the oensbi code, there are usually  a code style like this:
    .........................................................
    **sbi_hart_set_trap_info(scratch, trap);**
    //build in assemble language.
    ...............................
    ...............................
    **sbi_hart_set_trap_info(scratch, NULL);**

for example:
 **29                 sbi_hart_set_trap_info(scratch, trap);                        \**
 30                 asm volatile(                                                 \
 31                         "csrrs %0, " STR(CSR_MSTATUS) ", %3\n"                \                                                                                                            
 32                         ".option push\n"                                      \
 33                         ".option norvc\n"                                     \
 34                         #insn " %1, %2\n"                                     \
 35                         ".option pop\n"                                       \
 36                         "csrw " STR(CSR_MSTATUS) ", %0"                       \
 37                 ¦   : "+&r"(__mstatus), "=&r"(val)                            \
 38                 ¦   : "m"(*addr), "r"(MSTATUS_MPRV));                         \
 **39                 sbi_hart_set_trap_info(scratch, NULL);                        \**
 40                 return val;                                                   \

    so,what is the use of symmetry use  of this  api ?

thank you!

avpatel, 2020-08-11:
This was for capturing trap details for M-mode to M-mode trap while doing unprivileged access.

This function does not exist anymore because we have new way to capture trap details for M-mode to M-mode trap.

Regards,
Anup

13824125580, 2020-08-11:
from which version change to the new way？@avpatel

avpatel, 2020-08-11:
This change happened between v0.6 and v0.7 release

Regards,
Anup

avpatel, 2020-10-21:
I assume this is resolved hence closing the issue.

closed 2020-10-21 by avpatel