riscv-software-src/opensbi #180: Which is the easiest way to boot linux with opensbi

dskwe, 2020-10-18:
I found several ways to boot linux with opensbi in a new platform:
a) specific fsbl->opensbi->linux
b) specific fsbl->opensbi->u-boot->linux
c) u-boot-spl->opensbi->u-boot->linux

I wonder which way is the easiest. I prefer a) , use opensbi firmware FW_PAYLOAD and set  linux kernel image as the payload. In this way, I don’t need to care about u-boot related code. The work will be focused on opensbi and linux. 
But it seems c) is the mainstream boot flow?

dskwe, 2020-10-18:
Are there some examples for a)? 
Reference links would be helpful, thanks.

avpatel, 2020-10-19:
For a), usually FPGA platforms use this approach. You can look at docs/platform/fpga-ariane.md OR docs/platform/shakti_cclass.md. The biggest issue with a) is that we have to recreate fw_payload.bin whenever kernel image changes.

For b) and c), mostly real world SoC tend to use these flows. We have U-Boot (or some other open source bootloader) which provides advance booting methods (such as network booting). We can load kernel image from SD card OR Flash OR TFTP without changing firmware or bootloader.

The c) approach is most flexible because we can upgrade any image in the boot flow without changing other images.

Regards,
Anup

dskwe, 2020-10-19:
Yes, I'm doing my work on a FPGA, and want to use the a) approach.
What about platform/sifive_fu540? I have read linux and opensbi code for sifive_fu640, and would like to do some work based on the mainline of these two git repositories. 
On your suggestion, the FPGA platform such as fpga-ariane or shakti would be a better example for me?

dskwe, 2020-10-19:
> For a), usually FPGA platforms use this approach. You can look at docs/platform/fpga-ariane.md OR docs/platform/shakti_cclass.md. The biggest issue with a) is that we have to recreate fw_payload.bin [...]

Yes, I'm doing my work on a FPGA, and want to use the a) approach.
What about platform/sifive_fu540? I have read linux and opensbi code for sifive_fu640, and would like to do some work based on the mainline of these two git repositories.
On your suggestion, the FPGA platform such as fpga-ariane or shakti would be a better example for me?

avpatel, 2020-10-21:
I don't think there is FU640 from SiFive. Did you mean FU540?

The shakti_cclass support uses OpenSBI generic platform so I recommend you can take similar route but if you want to do things differently from OpenSBI generic platform then you can take fpga-ariane as reference.
 
Regards,
Anup

dskwe, 2020-10-23:
Yes, actually I mean  FU540. 

Thank you.

> I don't think there is FU640 from SiFive. Did you mean FU540?
> 
> The shakti_cclass support uses OpenSBI generic platform so I recommend you can take similar route but if you want to do things differently from OpenSBI generic platform then you can take fpga-ariane as reference.
> 
> Regards,
> Anup

avpatel, 2020-10-23:
I assume your issue is resolved. Can we close this issue?

Regards,
Anup

dskwe, 2020-10-24:
Sure.

closed 2020-10-24 by dskwe