riscv-software-src/opensbi #181: FU540 hart_count bug?

dskwe, 2020-10-29:
I Found below in the fu540/platform.c:

`const struct sbi_platform platform = {
	...
	.hart_count		= (FU540_HART_COUNT - 1),
	...
};

static struct clint_data clint = {
	...
	.hart_count = FU540_HART_COUNT,
	...
};`

And in lib/utils/sys/clint.c :
 `	/* Update timer hartid table */
	for (i = 0; i < clint->hart_count; i++)
		clint_timer_hartid2data[clint->first_hartid + i] = clint;`

Is there a bug for these two ".hart_count"?
It seems to be resigned both them (FU540_HART_COUNT-1).

avpatel, 2020-10-29:
It's not a bug.

Currently, on SiFive Unleashed we are not-using/disabling the E-core. There are two ways to achieve this:
1. Show less HARTs in "struct sbi_platform" and hart_index2id[] table will have not entry for E-core hartid
2. Show exact HARTs in "struct sbi_platform" and hart_index2id[] table will use -1UL as hartid for the E-core instead of actual hartid

The platform/sifive/fu540 takes approach#1 mentioned above whereas platform/generic takes approach#2. Going forward with OpenSBI domains being implemented, the approach#2 is much better and cleaner. If you want you can send patch for using approach#2 in platform/sifive/fu540.

Both platform/sifive/fu540 and platform/generic work fine on SiFive Unleashed board. The platform/sifive/fu540 is kind of redundant but we will let SiFive decide what to do with this directory.

Regards,
Anup

dskwe, 2020-11-23:
> It's not a bug.
> 
> Currently, on SiFive Unleashed we are not-using/disabling the E-core. There are two ways to achieve this:
> 
> 1. Show less HARTs in "struct sbi_platform" and hart_index2id[] table will have no entry for E-core hartid
> 2. Show exact HARTs in "struct sbi_platform" and hart_index2id[] table will use -1UL as hartid for the E-core instead of actual hartid
> 
> The platform/sifive/fu540 takes approach#1 mentioned above whereas platform/generic takes approach#2. Going forward with OpenSBI domains being implemented, the approach#2 is much better and cleaner. If you want you can send patch for using approach#2 in platform/sifive/fu540.
> 
> Both platform/sifive/fu540 and platform/generic work fine on SiFive Unleashed board. The platform/sifive/fu540 is kind of redundant but we will let SiFive decide what to do with this directory.
> 
> Regards,
> Anup

Thank you!

closed 2020-11-23 by dskwe