riscv-software-src/opensbi #183: how to pass the command with fw_payload.elf

zhuzhzh, 2020-11-28:
I follows the command to boot linux with fw_payload.elf. With this, I think vmlinux in included in fw_pyaload.elf, so -kernel is not used. but when -kernel is not used, -append is not allowed. 

```
qemu-system-riscv64 -M virt -m 256M -nographic -bios opensbi/build/platform/generic/firmware/fw_payload.elf -drive file=./rootfs.img,format=raw,id=hd0  -device virtio-blk-device,drive=hd0 -append "root=/dv/vda rw console=ttyS0"

Without -append, the boot will error out with the following error.

```
[    0.272843] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    0.273070] Please append a correct "root=" boot option; here are the available partitions:
[    0.273442] fe00         1048576 vda 
[    0.273466]  driver: virtio_blk
[    0.273773] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    0.274122] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.4.0 #1
[    0.274310] Call Trace:
[    0.274468] [<ffffffe000037ac4>] walk_stackframe+0x0/0xa6
[    0.274625] [<ffffffe000037c3a>] show_stack+0x2a/0x34
[    0.274793] [<ffffffe000621d4e>] dump_stack+0x6e/0x88
[    0.274925] [<ffffffe00003cb86>] panic+0xdc/0x246
[    0.275059] [<ffffffe000000f7a>] mount_block_root+0x184/0x218
[    0.275189] [<ffffffe00000111c>] mount_root+0x10e/0x124
[    0.275308] [<ffffffe000001244>] prepare_namespace+0x112/0x15c
[    0.275438] [<ffffffe000000c10>] kernel_init_freeable+0x188/0x1a4
[    0.275573] [<ffffffe00063662a>] kernel_init+0x12/0xf0
[    0.275690] [<ffffffe000036818>] ret_from_exception+0x0/0xc
[    0.276288] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0) ]---
QEMU: Terminated

```

How can I pass "root=" to kernel with fw_payload.elf?

alistair23, 2020-11-30:
You should use fw_jump.elf instead and then specify a kernel and -append when starting QEMU.

avpatel, 2020-12-05:
If you have to use fw_payload.elf then pass it using "-kernel" parameter instead of "-bios" parameter.

Regards,
Anup