riscv-software-src/opensbi #184: How can I disable domain completely?

dayeol, 2020-12-02:
Hello,

I have a question about the recent domain support because it broke the compatibility of our experimental security monitor library.
We've been using PMP entries for isolating Keystone enclave, which is being ported to OpenSBI.
I found that our PMP mechanism stopped working after merging a recent branch because PMP registers were somehow overwritten by OpenSBI.

I tried to clean the entire PMP registers in platform_final_init, but when there're multiple harts, only the booting hart is getting initialized.

Is there a way I can get away from the domain? or should I change keystone to use the domain instead?
Our firmware dynamically changes the PMP entries in order to control the memory visibility.
Also, we're planning to introduce more sophisticated PMP management for secure memory sharing and etc.
Thus, it'd be really hard to customize the current domain implementation.

Could you advise on this, please? Thank you!

dayeol, 2020-12-02:
updates:
- I managed to clean PMP registers for warm-booted harts. I thought that platform's final_init is only called by the cold-boot hart, but that was not true.
- Thus, I temporarily fixed the issue, but still, I have a question: am I expected to stick with the domain interface, or is it fine to have my own PMP mechanism? 
Thanks a lot!

avpatel, 2020-12-03:
We suggest Keystone enclave project to use OpenSBI domain interface. The OpenSBI domain is just an abstraction having all the details required to configure PMP, IOPMP, SiFive shield and other HW security features. We will keep enriching the OpenSBI domain implementation for various HW security features that show-up in future so it will be good for Keystone enclave project to depend on OpenSBI domain instead of directly programming HW security features.

The OpenSBI domain already has this notion of possible HARTs which the set of HARTs possible in this domain. A HART X can possible in multiple OpenSBI domains but at any point in time HART X is assigned exactly one OpenSBI domain. We need to add more OpenSBI domain APIs to allow HART X to switch to a different OpenSBI domain and re-configure various HW security features. These new OpenSBI domain APIs can help keystone enclave project.

Regards,
Anup

avpatel, 2021-01-18:
Can this be closed now ?

closed 2021-01-18 by dayeol