riscv-software-src/opensbi #185: OpenSBI service in HSS hangs

wojteklewandowski-pl, 2020-12-03:
Hello, 
I am trying to boot Linux using https://github.com/polarfire-soc/polarfire-soc-buildroot-sdk repository. What I want to change is the boot address. I would like to boot Linux from DDR connected to the process via FPGA. The address of this DDR is 0x20_0000_0000. I think I have changed it everywhere I had to. 
I am aware that it may be not exactly the right place to ask this question but I think that it hangs in the openSBI service in the HSS, where it should start U-boot. I have added some prints and it looks like that:
```
[11.172343] HSS_Boot_PMPSetupHandler(): Hart1 setup complete
HSS_OpenSBI_IPIHandler(): fw_size 1c214
HSS_OpenSBI_IPIHandler(): hss_start 20220000 1c214 
HSS_OpenSBI_IPIHandler(): p_extended_buffer 2030000028 
HSS_OpenSBI_Setup(): MTVEC switching from 20220230 to 20220100
```
I also had once the error like that:
`sbi_emulate_csr_read: hartid1: invalid csr_num=0x42`

I am looking for some advice how to debug it and find the source of the problem.

Regards
WL

avpatel, 2020-12-03:
There is already a fix on OpenSBI mailing to avoid accessing N-extension (i.e. User Trap Handling and Setup) CSRs.

Regards,
Anup

wojteklewandowski-pl, 2020-12-04:
Thank you Anup, for the response. So I can assume that there is no openSBI updated in the polarfire-soc-buildroot-sdk. I will update it by myself. Could you tell me what exactly may be a reason for this invalid csr_num? What causes that?

avpatel, 2020-12-05:
The invalid CSR error is caused by sbi_hart_switch_mode() function when next_mode == u-mode.

This is now fixed in latest OpenSBI repo so please pull the latest OpenSBI.

Regards,
Anup

avpatel, 2020-12-19:
The fix is already merged in latest OpenSBI.

Can you check and close this issue?

Regards,
Anup

wojteklewandowski-pl, 2020-12-21:
Hi Anum,
I will, but I am using the HSS application. Last time I have checked, it has older version of OpenSBI, and the older version of its service which uses the OpenSBI, so the update of openSBI requires an update of the service. I will do that and let you know if there is a problem. I am closing it for now. In case of problems I will reopen

wojteklewandowski-pl, 2021-01-11:
Hi Anup, 
I am reopening, but it may not exactly be an issue, but I am looking for some advice. I am trying to boot baremetal application from DDR using HSS (just like Microsemi did on this videos https://www.youtube.com/watch?v=CvhgPlVuV30  https://www.youtube.com/watch?v=H__COxfkb18), but I would like to do it from the DDR connected to the Fabric of PolarFire SoC. The DDR is mapped from 0x60000000 to 0x7FFFFFFF via FIC interface. I am getting the sbi_trap_error on mmuart1:

`[9.142917] IPI_ConsumeIntent(): ::mscgen: E51->U54_1 IPI_MSG_PMP_SETUP 1 0 0
[9.151337] HSS_Boot_PMPSetupHandler(): Hart1 setup complete
[9.158281] IPI_Send(): ::mscgen: U54_1->E51 IPI_MSG_ACK_COMPLETE 1 1 0
[9.166180] HSS_U54_HandleIPI():  ipi_interrupts: 1 (1)
[9.172950] IPI_DebugDumpStats(): message_allocs:   0
[9.179547] IPI_DebugDumpStats(): message_delivers: 0
[9.186144] IPI_DebugDumpStats(): message_frees:    0
[9.192741] IPI_DebugDumpStats(): consume_intents:  1
[9.199338] IPI_DebugDumpStats(): ipi_sends:        1
[9.566944] IPI_ConsumeIntent(): ::mscgen: E51->U54_1 IPI_MSG_OPENSBI_INIT 5 3 7a000028
[9.576232] IPI_Send(): ::mscgen: U54_1->E51 IPI_MSG_ACK_COMPLETE 5 1 0
HSS_OpenSBI_Setup(): MTVEC switching from 20220230 to 20220100

OpenSBI v0.6
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name          : Microchip PolarFire SoC
Platform HART Features : RV64ACDFIMSU
Platform Max HARTs     : 5
Current Hart           : 1
Firmware Base          : 0x20220000
Firmware Size          : 102 KB
Runtime SBI Version    : 0.2

MIDELEG : 0x0000000000000222
MEDELEG : 0x000000000000b109
PMP0    : 0x0000000020220000-0x000000002023ffff (A)
PMP1    : 0x0000000000000000-0x0000007fffffffff (A,R,W,X)
sbi_trap_error: hart1: illegal instruction handler failed (error -2)
sbi_trap_error: hart1: mcause=0x0000000000000002 mtval=0x00000000ffffffff
sbi_trap_error: hart1: mepc=0x0000000060200008 mstatus=0x8000000a00007800
sbi_trap_error: hart1: ra=0x000000002022ecda sp=0x0000000008019370
sbi_trap_error: hart1: gp=0x0000000000000000 tp=0x0000000008016f40
sbi_trap_error: hart1: s0=0x00000000080193c0 s1=0x8000000a00007800
sbi_trap_error: hart1: a0=0x0000000000000001 a1=0x0000000000000000
sbi_trap_error: hart1: a2=0x0000000060200000 a3=0x0000000000000001
sbi_trap_error: hart1: a4=0x0000000000000000 a5=0x0000000000000001
sbi_trap_error: hart1: a6=0x0000000000000010 a7=0x0000000000000010
sbi_trap_error: hart1: s2=0x0000000000000003 s3=0x8000000a00007800
sbi_trap_error: hart1: s4=0x0000000060200000 s5=0x0000000000000000
sbi_trap_error: hart1: s6=0x0000000000000001 s7=0x0000000000000000
sbi_trap_error: hart1: s8=0x0000000020235588 s9=0x0000000008000f88
sbi_trap_error: hart1: s10=0x00000000202357f8 s11=0x0000000000000000
sbi_trap_error: hart1: t0=0x0000000000000000 t1=0x0000000020227802
sbi_trap_error: hart1: t2=0x0000000000000001 t3=0x0000000000000027
sbi_trap_error: hart1: t4=0x0000000000000010 t5=0x0000000000000000
sbi_trap_error: hart1: t6=0x00000000080192a0`

Could you please let me know what may be a reason of that and what this crash excacly means?

wojteklewandowski-pl, 2021-01-14:
In addition - This sbi_trap_error appears after 

	`__asm__ __volatile__("mret" : : "r"(a0), "r"(a1));`

from  sbi_hart_switch_mode function in sbi_hart.c

Regards
WL

avpatel, 2021-01-14:
OpenSBI jumped into next booting stage in U-Boot and your next booting stage executed some illegal instruction.

This does not seem to be related to OpenSBI.

Regards,
Anup