riscv-software-src/opensbi #187: [K210] What is the current status of K210 support?I didn't make it work.

balanceTWK, 2020-12-18:
👀...

nylon7, 2021-01-22:
Could you have shown your error message?

hzcx998, 2022-04-25:
you can try to use rustsbi, it support k210 very well!

z3cko, 2023-07-08:
@hzcx998 do you have experience with rustsbi and k210? Do you mind sharing?

hzcx998, 2023-07-09:
> 

Yes, when i use rustsbi, it support mmu so good, but different with opensbi, it would not enable fpu auto, you need enable fpu by yourself.

z3cko, 2023-07-16:
@hzcx998 can you explain more in depth? Do you have any examples or details? I understand that rustsbi helps in running MMU linux (can you confirm?)

disdi, 2024-10-14:
can opensbi support nommu linux on [risv32 qemu target  ](https://github.com/buildroot/buildroot/blob/master/configs/qemu_riscv32_nommu_virt_defconfig) ?

avpatel, 2024-10-15:
NoMMU Linux runs in M-mode so I don't think OpenSBI can be useful in this case.

We will be soon dropping the kendryte/k210 support from OpenSBI because Linux NoMMU doesn't need it.

Regards,
Anup

SiFiveHolland, 2024-10-15:
NOMMU Linux runs in S-mode as of [f862bbf4cdca](https://git.kernel.org/torvalds/c/f862bbf4cdca696ef3073c5cf3d340b778a3e42a) (v6.10).

avpatel, 2024-10-16:
Latest Linux now supports S-mode NoMMU but I still don't see any users of kendryte/k210 OpenSBI platform.