riscv-software-src/opensbi #188: Additional scratch interface for getting offset of a certain owner?

dayeol, 2020-12-28:
My custom IPI handler needs synchronization over all cores just like how TLB IPI is handled.
It's PMP IPI I implemented which synchronizes a certain PMP region over all cores.
However, it may end up with a deadlock if one core is synchronizing TLB IPI while the other is synchronizing PMP IPI.

An easy fix I came up with is my PMP IPI sync to consume TLB FIFO because a PMP IPI request will always flush the entire TLB.
However, the relevant offsets are currently static variables within [sbi_tlb.c](https://github.com/riscv/opensbi/blob/a04c46506a6e5bd7425bd63430645a6c85145b10/lib/sbi/sbi_tlb.c#L26)
I had to make it non-static in order to get the offset of the FIFO.

However, it would be much better if there's a way to get the offset from the outside.
Would there be any recommendation on how to implement this?
I'm happy to contribute if you give some direction.

e.g., I do see `char* owner` being passed when you allocate an offset in the scratch, but it's never been used.
Can we use this variable to get the offset? (like `sbi_scratch_get_offset_by_owner("TLB_FIFO")`)

avpatel, 2020-12-28:
How about we extend our TLB implementation to allow users set a callback function "struct sbi_tlb_info" ?
This will help you write your own custom TLB flush handler which will do both PMP update and flush entire TLB.

Regards,
Anup

dayeol, 2020-12-28:
Thanks, I'll try that and send out the patch.

avpatel, 2021-01-18:
Can this be closed now ?

dayeol, 2021-01-18:
The issue has been resolved, Thank you!

closed 2021-01-18 by dayeol