riscv-software-src/opensbi #208: No longer builds after enabling PIC by default

repnop, 2021-04-28:
Hi, I just updated my submodule for OpenSBI and it no longer builders and exits with the following error:
```
/usr/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: -pie not supported
collect2: error: ld returned 1 exit status
make: *** [Makefile:351: /home/repnop/dev/rust/vanadinite/submodules/opensbi/build/platform/generic/firmware/payloads/test.elf] Error 1
make: *** Waiting for unfinished jobs....
 ELF       platform/generic/firmware/fw_dynamic.elf
/usr/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: -pie not supported
collect2: error: ld returned 1 exit status
make: *** [Makefile:351: /home/repnop/dev/rust/vanadinite/submodules/opensbi/build/platform/generic/firmware/fw_dynamic.elf] Error 1
Error: command `make PLATFORM=generic FW_PAYLOAD_PATH=../../src/target/riscv64gc-unknown-none-elf/release/vanadinite.bin` failed, exit status: 2
```

Confirmed that it correctly builds after checking out to before the commit that enabled PIC by default.

avpatel, 2021-04-29:
Looks like, a toolchain issue.

I think the "-pie" is supported by latest toolchain.

Regards,
Anup

repnop, 2021-04-29:
I had a feeling that was the case but wanted to double check, thanks. Looks like the AUR RISC-V binutils package is sorely out of date. :(

closed 2021-04-29 by repnop

repnop, 2021-05-02:
Hm, I just installed the toolchain from the Github repo and I'm still receiving that error, is that expected?
```
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: -pie not supported
collect2: error: ld returned 1 exit status
make: *** [Makefile:351: /home/repnop/dev/rust/vanadinite/submodules/opensbi/build/platform/generic/firmware/payloads/test.elf] Error 1
make: *** Waiting for unfinished jobs....
 ELF       platform/generic/firmware/fw_dynamic.elf
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: -pie not supported
collect2: error: ld returned 1 exit status
make: *** [Makefile:351: /home/repnop/dev/rust/vanadinite/submodules/opensbi/build/platform/generic/firmware/fw_dynamic.elf] Error 1
Error: command `make PLATFORM=generic FW_PAYLOAD_PATH=../../src/target/riscv64gc-unknown-none-elf/release/vanadinite.bin` failed, exit status: 2
```
(10.2.0 instead of 9.2.0 this time)

avpatel, 2021-05-02:
Most of us are using Linux toolchain complied using risc-gnu-toolchain repo.

I guess that toolchain has PIE enabled by default.

Regards,
Anup