riscv-software-src/opensbi #219: beaglev: no OpenSBI banner and info displayed?

pdp7, 2021-06-13:
I am trying to switch beaglev_defconfig in buildroot over to upstream OpenSBI and generic platform
```
CROSS_COMPILE=/home/pdp7/dev/buildroot/output/host/bin/riscv64-buildroot-linux-uclibc- \
PLATFORM=generic \
FW_PAYLOAD_PATH="/home/pdp7/dev/buildroot/output/images/u-boot.bin" \ 
FW_FDT_PATH="/home/pdp7/dev/buildroot/output/images/u-boot.dtb" \
/usr/bin/make -j9 -C /home/pdp7/dev/buildroot/output/build/opensbi-0.9
```
It is building the fw_payload with u-boot.bin and u-boot.dtb.

It appears to boot ok but I do not see the OpenSBI banner and info like is normally displayed.  Instead, it just shows `bootloader.`
```
bootloader version:buildroot-f5b3e894a2a9a4c13d087d3210a694285a803eb0
ddr 0x00000000, 1M test
ddr 0x00100000, 2M test
DDR clk 2133M,Version: buildroot-c0839f25246d9e308c23498d344ca13d8a7ad6ed
0 crc flash: ddc982dc, crc ddr: ddc982dc
crc check PASSED

bootloader.

U-Boot 2021.04 (Jun 13 2021 - 12:48:13 -0700)

CPU:   rv64imafdc
DRAM:  8 GiB
MMC:   sdio0@10000000: 0, sdio1@10010000: 1
Net:   dwmac.10020000
Autoboot in 2 seconds
MMC CD is 0x1, force to True.
MMC CD is 0x1, force to True.
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
183 bytes read in 6 ms (29.3 KiB/s)
1:      linux
Retrieving file: /boot/Image
19701248 bytes read in 4083 ms (4.6 MiB/s)
append: console=ttyS0,115200 earlyprintk root=PARTUUID=0fef845a-c6e1-45bc-82f7-002fa720f958 rootwait
Retrieving file: /boot/jh7100-beaglev-starlight.dtb
17366 bytes read in 10 ms (1.7 MiB/s)
Moving Image from 0x84000000 to 0x80200000, end=8151c000
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 00000000ffff8000, end 00000000fffff3d5 ... OK

Starting kernel ...

Linux version 5.13.0-rc2-starlight-00046-g927379f85f8c-dirty (atish@jedi-01) (riscv64-unknown-linux-gnu-gcc (GCC) 10.2.0, GNU ld (GNU Binut1
OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
Machine model: BeagleV Starlight Beta
```
@lbmeng @atishp04 @mcd500 @avpatel any idea what I am doing wrong?

alistair23, 2021-06-13:
I saw this with the v0.9 release, the lastest master printed the OpenSBI banner though.

lbmeng, 2021-06-13:
It seems you are using OpenSBI v0.9, which does not have the patch for BeagleV serial port:

```
opensbi-0.9.tar.gz: OK (sha256: 60f995cb3cd03e3cf5e649194d3395d0fe67499fd960a36cf7058a4efde686f0)
>>> opensbi 0.9 Extracting
gzip -d -c /home/pdp7/dev/buildroot/dl/opensbi/opensbi-0.9.tar.gz | tar --strip-components=1 -C /home/pdp7/dev/buildroot/output/build/opensbi-0.9   -xf -
>>> opensbi 0.9 Patching

Applying 0001-Makefile-Don-t-specify-mabi-or-march.patch using patch: 
patching file Makefile
Hunk #1 succeeded at 206 with fuzz 1 (offset 61 lines).
Hunk #2 succeeded at 219 (offset 62 lines).

Applying 0002-Makefile-unconditionally-disable-SSP.patch using patch: 
patching file Makefile
Hunk #1 succeeded at 203 with fuzz 1.
Hunk #2 succeeded at 216 with fuzz 1 (offset -1 lines).
Hunk #3 succeeded at 234 (offset -2 lines).
>>> opensbi 0.9 Configuring
>>> opensbi 0.9 Building
```

pdp7, 2021-06-14:
@alistair23 @lbmeng thank you!  yes, I should have been using master.

closed 2021-06-14 by pdp7