riscv-software-src/opensbi #220: undefined reference to `__stack_chk_guard'

ramprakashb, 2021-06-16:
I am trying to emulate a RISCV Virt board and I have followed the toolchain setup according to the bootlin slides available here:

https://bootlin.com/pub/conferences/2021/fosdem/opdenacker-embedded-linux-45minutes-riscv/opdenacker-embedded-linux-45minutes-riscv.pdf
https://www.youtube.com/watch?v=cIkTh3Xp3dA

- I have generated the RISCV musl toolchain using buildroot.
- Setup QEMU hardware emulator.
- Built the UBoot binaries using Uboot 2021.01 sources in S mode. (qemu-riscv64_smode_defconfig)
- Followed this by setting up OpenSBI:

But when I try to build OpenSBI using command:
make PLATFORM=generic FW_PAYLOAD_PATH=../u-boot-2021.01/u-boot.bin 

I get the following error. Can someone help me in fixing it?

/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_init.o): in function `sbi_init':
/home/ram/Documents/opensbi/lib/sbi/sbi_init.c:290: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_ipi.o): in function `sbi_ipi_send_many':
/home/ram/Documents/opensbi/lib/sbi/sbi_ipi.c:77: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/sbi_ipi.c:90: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_tlb.o): in function `sbi_tlb_entry_process':
/home/ram/Documents/opensbi/lib/sbi/sbi_tlb.c:212: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_tlb.o): in function `sbi_tlb_process':
/home/ram/Documents/opensbi/lib/sbi/sbi_tlb.c:241: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/sbi_tlb.c:244: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_tlb.o): in function `sbi_tlb_process_count':
/home/ram/Documents/opensbi/lib/sbi/sbi_tlb.c:222: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_trap.o): in function `sbi_trap_handler':
/home/ram/Documents/opensbi/lib/sbi/sbi_trap.c:213: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/sbi_trap.c:261: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(platform.o): in function `fw_platform_init':
/home/ram/Documents/opensbi/platform/generic/platform.c:73: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/platform/generic/platform.c:118: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(riscv_asm.o): in function `misa_string':
/home/ram/Documents/opensbi/lib/sbi/riscv_asm.c:53: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/riscv_asm.c:70: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_console.o): in function `prints':
/home/ram/Documents/opensbi/lib/sbi/sbi_console.c:116: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_console.o): in function `printi':
/home/ram/Documents/opensbi/lib/sbi/sbi_console.c:168: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_console.o): in function `print':
/home/ram/Documents/opensbi/lib/sbi/sbi_console.c:178: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/sbi_console.c:325: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/sbi_console.c:288: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_console.o): in function `sbi_platform_console_getc':
/home/ram/Documents/opensbi/include/sbi/sbi_platform.h:522: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_console.o): in function `sbi_sprintf':
/home/ram/Documents/opensbi/lib/sbi/sbi_console.c:342: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/sbi_console.c:342: undefined reference to `__stack_chk_guard'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(sbi_console.o): in function `sbi_snprintf':
/home/ram/Documents/opensbi/lib/sbi/sbi_console.c:354: undefined reference to `__stack_chk_fail'
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/lib/sbi/sbi_console.c:354: undefined reference to `__stack_chk_guard'
[... cut ...]
/home/ram/Documents/toolchain/riscv64-buildroot-linux-musl_sdk-buildroot/bin/../lib/gcc/riscv64-buildroot-linux-musl/9.3.0/../../../../riscv64-buildroot-linux-musl/bin/ld: /home/ram/Documents/opensbi/build/platform/generic/lib/libplatsbi.a(fdt_wip.o): in function `fdt_nop_node':
/home/ram/Documents/opensbi/lib/utils/libfdt/fdt_wip.c:93: undefined reference to `__stack_chk_fail'
collect2: error: ld returned 1 exit status
Makefile:317: recipe for target '/home/ram/Documents/opensbi/build/platform/generic/firmware/fw_dynamic.elf' failed
make: *** [/home/ram/Documents/opensbi/build/platform/generic/firmware/fw_dynamic.elf] Error 1

Cuicunhao, 2021-12-08:
Hello, I met the same problem as you. Have you solved it？

riggy2013, 2022-03-01:
add CFLAGS += -fno-stack-protector in Makefile can solve the issue.