riscv-software-src/opensbi #233: How to get my hartid in SMP？

hzcx998, 2021-12-12:
In opensbi, OS running in supervisor mode，I can't get hartid from mhartid, I just wanna a function to get hartid like `shartid`，even though it is read-only, that works! How can I do that？

repnop, 2021-12-12:
The hart ID will be passed to you in register `a0` when OpenSBI transfers execution to your kernel, in both cases of initial boot and whenever you use the `sbi_hart_start` call, you can then use a hart-local data structure to store that information long term.

hzcx998, 2021-12-12:
@repnop Thank you for your reply, I know that I can store it in `gp` or `tp`, but they are not safe and can be modified in user mode. I tried to fill in different stvals when initializing the trap to indicate different cpus. When getting it, just make an address judgment. It works very well, but it is not elegant enough.

hzcx998, 2021-12-12:
@repnop This is my not-so-good-looking code.
```c

/**
 * Within SBI, we can't read mhartid, so I try to use `trap entry` to see who am I.
 */
PUBLIC Uint HAL_CoreGetId(void)
{
    Addr trapEntry = ReadCSR(stvec);

    if (trapEntry == (Addr)&TrapEntry0)
    {
        return 0;
    }
    else if (trapEntry == (Addr)&TrapEntry1)
    {
        return 1;    
    }
    else if (trapEntry == (Addr)&TrapEntry2)
    {
        return 2;    
    }
    else if (trapEntry == (Addr)&TrapEntry3)
    {
        return 3;    
    }
    /* should never be here */
    while (1);
}
```

repnop, 2021-12-12:
> but they are not safe and can be modified in user mode.
You need to save and restore all of your kernel state before switching to user-mode regardless, so storing it in `tp`, for instance, isn't really a problem assuming that you don't have any `thread_local`s that will try to use the register for its intended purpose. (I would _highly_ advise using `gp` since your kernel code will likely want to use that) But I would also advise that you look into either adding `thread_local`s that function as core-local data using `tp`, or you can use a pointer to a data structure in `sscratch` to store information like that as well.

hzcx998, 2021-12-12:
@repnop I once considered the `sscratch` register, but it was used by me for other purposes. I think about how to make it better, thank you for your reply!