riscv-software-src/opensbi #234: memset / memcpy builds into libsbi.a

eenurkka, 2021-12-17:
See the comment related to the patch:
https://github.com/riscv-software-src/opensbi/commit/d249d6544c3cbf6a817e2fac40b4be0e006ba965#comments

Linking against the libsbi.a causes memset / memcpy to be located in the libsbi.a in some cases. This is probably not what is wanted; strong memset / memcpy's will be ignored and the ones in libsbi.a will be used. It becomes a mixed library, although probably original memset and memcpy should be placed in their original context.

Without that patch all is fine.

avpatel, 2021-12-17:
Without that patch compilation of OpenSBI firmware fails for "-Os" so it was best to have weak aliases of memcpy() and memset() so that external firmware (using OpenSBI as library) can easily override it.

eenurkka, 2022-01-04:
This looks like fixed, thanks!

closed 2022-01-04 by eenurkka