riscv-software-src/opensbi #239: v 1.0 not compatible with fu540 HiFive unleashed ?

klaus4, 2022-01-14:
a 1st attempt failed to port HiFiveUnleashed errata-bugfixes to OpenSBI v 1.0,
tested on the device 
https://reviews.freebsd.org/D33845
while I have an estimation of the root cause  errata I am not aware of all the details .

jrtc27, 2022-01-14:
Not a bug in OpenSBI. FreeBSD doesn't currently honour no-map FDT entries, which is what OpenSBI adds to tell the OS not to map the PMP-protected range in order to avoid the erratum. FreeBSD's OpenSBI patch is a hack until such time as the kernel is fixed to honour the entry.

klaus4, 2022-01-14:
ah, thanks for clarifying! the other question is why the hack worked before v 1.0, but O.K., not an OpenSBI bug, so closing this.

closed 2022-01-14 by klaus4