riscv-software-src/opensbi #240: opensbi early debugging

luca-valente, 2022-01-18:
Hi,
really sorry to bother you. I am trying to run openSBI on my RISC-V soc, which is a modified version of the standard [ariane soc](https://github.com/openhwgroup/cva6#corev-apu-fpga-emulation): more IPs and different memory map. Anyway, I did not modify ariane, the plic, the clint and the uart. I am targeting fpga-emulation on a xilinx vcu118.

I am trying to run the [dummy test](https://github.com/riscv-software-src/opensbi/blob/master/firmware/payloads/test_main.c#L45). I started by doing a [fork of opensbi](https://github.com/AlSaqr-platform/opensbi/tree/master/platform/fpga/alsaqr). 

On FPGA, I load the code with openocd and then the core jumps to `0x80000000`. When I set `PLATFORM=fpga/alsaqr`, opensbi works, also the dummy test is able to print.

When I set `PLATFORM=generic`, and link the device binary tree by writing the `dtb` address to `$a1`, it does not print anything. The code I execute before jumping to opensbi is the following (`fw_payload.elf` code is preloaded in the main memory at `0x80000000`):

```
int main() {
    init_uart(30000000, 115200);
    print_uart("Hello World!\r\n");

            __asm__ volatile(
                "csrr a0, mhartid;"
                "la a1, device_tree;"
                "li s0, 0x80000000;"
                "jr s0"
            );
}
```

My guess is that I am not properly generating the `dtb`. As SBI+dummy test is just 250kb of binary, I simulated the two opensbi configurations also in Questasim to see what the differences are, but without much success.

I hope this is the right place to get some support: is there any flag to ease early opensbi debugging? Like enabling the uart_8250 also when reading the `dtb`? Or is it impossible? I can see why it wouldn't make much sense honestly.

Thank you very much,
Luca

avpatel, 2022-01-21:
Can you point me to the DTS source of the DTB being passed to OpenSBI?

Regards,
Anup

luca-valente, 2022-01-21:
Hi @avpatel ,
thanks for the answer!

The dts is [this one](https://github.com/minho-pulp/cva6/blob/dtbdebug/corev_apu/bootrom/ariane.dts).

Thank you very much,
Luca

avpatel, 2022-01-21:
The compatible string "ns16750" of uart@10000000 node does not match the set of compatible strings supported by OpenSBI uart8250 driver.

Can you try ns16550 compatible string ?

Regards,
Anup

luca-valente, 2022-01-21:
Sure. 
I just launched the test.
Thanks again,
Luca

luca-valente, 2022-01-21:
By changing ns16750 to 16550, I get some uart output!
```
# [UART]: 
#                                                                                                                                                                                                                                                                
# [UART]: OpenSBI v0.9-207-g909d2a4
#                                                                                                                                                                                                                                       
# [UART]:    ____                    _____ ____ _____
#                                                                                                                                                                                                                     
# [UART]:   / __ \                  / ____|  _ \_   _|
#                                                                                                                                                                                                                    
# [UART]:  | |  | |_ __   ___ _ __ | (___ | |_) || |
#                                                                                                                                                                                                                      
# [UART]:  | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
#                                                                                                                                                                                                                      
# [UART]:  | |__| | |_) |  __/ | | |____) | |_) || |_
#                                                                                                                                                                                                                     
# [UART]:   \____/| .__/ \___|_| |_|_____/|____/_____|
#                                                                                                                                                                                                                    
# [UART]:         | |
#                                                                                                                                                                                                                                                     
# [UART]:         |_|
#                                                                                                                                                                                                                                                     
# [UART]: 
#                                                                                                                                                                                                                                                                
# 
```
But it looks like it's still not able to complete the whole execution. Being a Questasim simulation, I have access to the core's traces. Should I look to something in particular?

Thanks,
Luca

luca-valente, 2022-01-21:
I jumped too early to the conclusion! It works, sorry for that.
If you agree, I can close the issue.
Thank you very much again.
Luca

atishp04, 2023-01-12:
Closing this issue due to lack of activity.

closed 2023-01-12 by atishp04