riscv-software-src/opensbi #244: How to boot harts that's not boot hart?

retrhelo, 2022-02-08:
I'm developing a tiny kernel based on OpenSBI, and debugging it on QEMU. I use the flag `-smp 2` to have multiple harts. However, I can't find a proper way to "wake up" hart 1 (I use hart 0 as the boot core in my kernel). I wonder what I can do to boot hart 1? My QEMU version is 5.2.0 with OpenSBI at v0.8.

repnop, 2022-02-08:
Note that you generally do not get the luxury of picking which hart will boot: OpenSBI runs a hart lottery during initialization that will select a "random" (decided by atomic memory accesses) which will be your boot hart, I _strongly_ recommend against hardcoding the assumption that hart 0 is always the boot hart, as that is rarely going to be the case on multi-hart hardware. To then boot the other harts, the SBI specification provides the Hart State Management (HSM) extensions which allows you to ask the SBI implementation to start running code at a particular physical address with an additional user-supplied parameter on the other harts. Details in the SBI specification [here](https://github.com/riscv-non-isa/riscv-sbi-doc/blob/ca300998d6800cc29b95d4e138739e55f9629c72/riscv-sbi.adoc#function-hart-start-fid-0).

retrhelo, 2022-02-09:
The problem seems to be solved by using sbi_hart_start(). Thank you guys for the help :D.

retrhelo, 2022-02-09:
> Note that you generally do not get the luxury of picking which hart will boot: OpenSBI runs a hart lottery during initialization that will select a "random" (decided by atomic memory accesses) which [...]

Assuming hart 0 as the default boot hart is surely a bad idea. I'll imrove the kernel later. Thanks for pointing this out.

closed 2022-02-09 by retrhelo