riscv-software-src/opensbi #263: Document all available syscalls

kallisti5, 2022-07-20:
I've been looking around for a bit, and haven't found a list of the available opensbi syscalls.

get_spec_version, get_impl_id, get_impl_version, etc.

If these are part of the RISCV ISA or something, OpenSBI should at minimum tell users where in the spec they are.

avpatel, 2022-07-20:
The SBI v1.0 specification was ratified few months back.
https://github.com/riscv-non-isa/riscv-sbi-doc/releases/tag/v1.0.0

kallisti5, 2022-07-20:
Thanks!  I just double checked and this isn't linked to anywhere in the README.md in this repo.

Maybe a link to "SBI Syscall documentation" under the Documentation section?

atishp04, 2023-01-12:
OpenSBI is always compliant with latest SBI specifications. IMO that should be sufficient.

closed 2023-01-12 by atishp04