riscv-software-src/opensbi #266: Question about domain and hart

gagachang, 2022-08-21:
Will you consider to register multiple domains into one hart?

Refer to the following code in `sbi_domain.c`, it seems each hart has only one domain, am I correct?

Consider the single hart system, it might has multiple domains, said trusted and untrusted domains.
And the hart switches between these domains in the runtime, just like world-switch in ARM TrustZone.

```
int sbi_domain_register(struct sbi_domain *dom,
			const struct sbi_hartmask *assign_mask)
{
        ......

	tdom = hartid_to_domain_table[i];
	if (tdom)
		sbi_hartmask_clear_hart(i,
				&tdom->assigned_harts);
	hartid_to_domain_table[i] = dom;              <------- overwrite existed domain tdom
	sbi_hartmask_set_hart(i, &dom->assigned_harts);

        ......
}
```

The example in `domain_support.md` assigns trusted and untrusted domains to different harts.
```
tdomain: trusted-domain {
         compatible = "opensbi,domain,instance";
         possible-harts = <&cpu0>;
         regions = <&tmem 0x7>, <&tuart 0x7>;
         boot-hart = <&cpu0>;
};

udomain: untrusted-domain {
         compatible = "opensbi,domain,instance";
         possible-harts = <&cpu1 &cpu2 &cpu3 &cpu4>;
         regions = <&tmem 0x0>, <&tuart 0x0>, <&allmem 0x7>;
};
```

avpatel, 2022-09-01:
Yes, eventually we want to support same HART to be in "possible-harts" list of multiple domain.

We also want to support changing current domain assigned to a HART as long as the HART is listed as a possible HART for the newly assigned domain as well.

Regards,
Anup

gagachang, 2022-10-08:
Thanks for reply, that would be more flexible.

closed 2022-10-08 by gagachang

gagachang, 2023-11-06:
Hello @avpatel 

> Yes, eventually we want to support same HART to be in "possible-harts" list of multiple domain.
> 
> We also want to support changing current domain assigned to a HART as long as the HART is listed as a possible HART for the newly assigned domain as well.
> 
Is there any update about this? Thanks!