riscv-software-src/opensbi #268: xilinx uartlite

HaogeL, 2022-09-07:
Hi,

If I want to use xilinx uartlite driver instead of uart8250, how should I modify **platform_console_init** in **platform.c** for my own platform? 

```C
/*
 * Initialize the platform console.
 */
static int platform_console_init(void)
{
	/* Example if the generic UART8250 driver is used */
	return uart8250_init(PLATFORM_UART_ADDR, PLATFORM_UART_INPUT_FREQ,
			     PLATFORM_UART_BAUDRATE, 0, 1, 0);
}
```

Thanks in advance!
Hao

alistair23, 2022-09-08:
OpenSBI will pick the driver from the device tree, so you should try and use that.

Otherwise you could just call the `xlnx_uartlite_init()` function

HaogeL, 2022-09-09:
> OpenSBI will pick the driver from the device tree, so you should try and use that.
> 
> Otherwise you could just call the `xlnx_uartlite_init()` function
Hi, thanks for you reply.
Do you mean to drive Xilinx Uartlite properly, I need to 
1. setup device tree correctly and
2. replace `uart8250_init` with `xlnx_uartlite_init` in platform.c ?

closed 2022-09-09 by HaogeL