riscv-software-src/opensbi #271: [Debug opensbi] Using qemu+gdb to debug opensbi

hptsf, 2022-10-04:
Hi,

building opensbi, and then starting system using qemu with -S -s params

in other terminal, riscv64_unknown_linux_gdb fw_payload.elf
target remote :1234 
b sbi_init
c
n
will get tips as below, and can't stop on next line
[image]

hptsf, 2022-10-08:
Anyone has any idea?

alistair23, 2022-10-09:
Have you tried `step` instead of `next`?

hptsf, 2022-10-10:
yes， step is ok， but is not what i want. Maybe step is too slow.

Dunky-Z, 2022-11-07:
Maybe you can check if the GCC you use to compile and the GDB you use to debug are the same version. For example, I may have multi versions of the toolchain, sometimes mixing them.

Compile with `riscv64-unknown-linux-gnu-gcc` and debug with `riscv64-unknown-linux-gnu-gcc` may not be able to debug normally.

lbmeng, 2023-07-22:
When you build the OpenSBI binary, you can pass "DEBUG=1" on the command line which generates debugger-friendly bits.